supermerill/SuperSlicer #2636: SuperSlicer laggy on fullscreen ultrawide

manynie, 2022-04-03:
### Version
2.3.57.12

### Operating system type + version
Windows 10 Home 21H2 (recently installed)

### Behavior
So i found a problem with SuperSlicer. It's lagging when i put it on whole screen, when the window is small or not fullscreen it works perfectly.

supermerill, 2022-04-03:
Didn't have the problem on 256°x1440.
And didn't notice anything on my 4k screen

manynie, 2022-04-04:
That's how it looks on video, i found out that the problem occurs when window is close to full size of screen.
I had the same problem on windows 11, came back to fresh installed windows 10 and it didn't solve this issue.
I think that the problem is not with resolution of screen but with screen ratio.

https://user-images.githubusercontent.com/102881252/161556794-a75354a7-7190-40bc-b054-04a81cdffc35.mp4

supermerill, 2022-04-04:
Did you use a 100% scaling on windows or 125, 150% ?

manynie, 2022-04-04:
I use 100%

manynie, 2022-04-04:
I changed screen resolution to 2560x1080 and the problem disappeared. But on 3440x1440 still lags. I tried also to change windows scaling but it didn't change anything.

supermerill, 2022-04-04:
weird. I wonder if the opengl driver you're using doesn't have a problem with ultrawide.
The ultra wide is configured as one screen or multiple 'virtual' screens? (if you maximize a window, does it take all the space or only a part)

rjl9, 2022-08-23:
I know this is a bit old, but I to am running 3440X1440 and experience the same issues. rtx3080m 5800x all drivers upt to date on win10. Dont know if you found a work around or not.

neophyl, 2022-08-23:
I can run any of the versions on my win 10 machine at 3440x1440.  No lag at all.  graphics is a rtx3060.  Driver version is 516.59 from June.  Haven't got round to updating yet. Scaling is normally 125% but changing it doesn't make any difference either.  

I quite often run both SuSi and PS at the same time, both full screen and neither has a problem on my setup.  The again I've got 64gb of ram to play with and a Ryzen 9 :)