supermerill/SuperSlicer #2637: FEATURE REQUEST: Profile Comparison (complete)

inspectionsbybob, 2022-04-03:
### Version
2.3.57.11

### Operating system type + version
Win10

### Behavior
Need a way to compare -ALL- of Profile/config to another

_Is this a new feature request?_
YES

Wanted to load a "Super Fast Profile" that someone had developed for a Prusa printer, but I have a different printer so all of the settings may not be appropriate, need a way to compare a complete PROJECT profile SET to another to quickly find the differences.

Would like to see an option/utility to identify/select two PROJECT FILES (3MF) and then compare them side by side and highlight in **RED** the differences  AND have the option to PRINT the results.  

Looked at "Configuration Snapshot" and it only compared a small portion of the complete (Print settings, filament settings, Printer Settings) profile.  Finding the rest of the differences means opening every page, scrolling, scrolling, and hoping I don't miss something

The profile is from CNCKitchen and is at: https://www.cnckitchen.com/s/Ender3-Extrafast_Profile.3mf    It was too Big to attach (kept saying not included)

neophyl, 2022-04-04:
Bob its not too big, it just needed to be zipped up.  File types here are restricted, similar to the Prusa forum.

supermerill, 2022-04-04:
isn't it already implemented in the 2.4?

Dx-Dx-Dx, 2022-04-04:
It is.
[image]
[image]

inspectionsbybob, 2022-04-04:
Again, Ragu, its in there, you just need to search (more)

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2022-10-03:
Done in current version

closed 2022-10-03 by inspectionsbybob