supermerill/SuperSlicer #2638: Incorrectly generated parameter "skirt_extrusion_width"

Hashan0, 2022-04-03:
### What happened?

If the parameter  "_skirt_extrusion_width_"  in  "_vendor\xxx.ini_ "  file is missing or empty, the value **130** instead of **130%** is generated.
[image]
[image]

### Project file & How to reproduce

[Desktop.zip](https://github.com/supermerill/SuperSlicer/files/8405379/Desktop.zip)

### Version

2.4.58.2

### Operating system

Windows 10

supermerill, 2022-04-04:
how did you get that? if i open the 2.4.58.2 without (or an empty) configuration directory, I get a 130%.

Hashan0, 2022-04-04:
The error occurs when the parameter "skirt_extrusion_width = " is entered in the config file (see "vendor\Basic.ini" line 347).
[image]
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/8408625/SuperSlicer.zip)

closed 2024-02-11 by supermerill