supermerill/SuperSlicer #2639: Bad quality with painting tool [2.4.58.2 Fishy edition]

schnello, 2022-04-04:
Hello.

I want only ask if this issue is know,

If i use the color painting tool with the latest test version the surface quality is bed ([Fishy edition](https://github.com/supermerill/SuperSlicer/releases/tag/2.4.58.2)
[files.zip](https://github.com/supermerill/SuperSlicer/files/8407795/files.zip)

If i do the same with the prusa all is fine

[image]

NoSvOrAx, 2022-05-07:
I'm having the same issue with the Fishy Edition.  Just tried it on the previous version, 2.4.58.1, and it doesn't have the problem.

[image]

supermerill, 2022-09-09:
awfully difficult to fix.
The probem where in the resolution. The paint is made on the raw polygons, and it's compared to the simplified polygon.
The threshold was too small, and it can't find the new lines anymore.

Tell me if artefacts are appearing again.

schnello, 2022-09-17:
Tested with the lastest nightly. 
left Superslicer, right PrusaSlicer
[image]

Nerobeus, 2022-10-03:
Still have troubles... See the test part from ERCF.

[image]

fract0id, 2022-11-23:
Just saw this, not sure it's still current. I had a similar issue, but it appeared in both PS and SS with the same project, but not when creating a new project in PS. I eventually figured out that it was caused by the "Slice resolution" parameter. PS defaults to 0, in SS I had 0.0125. Setting it to 0 fixed the issue for me, with no noticeable downsides.

MarkXYZ-Mtl, 2023-03-14:
@fract0id 
It had to adjust "Slice resolution" for  SS 2.5.59.2 (latest pre-release) and 2.4.58.5.
Been looking for a while for a solution!
Thanks

closed 2024-02-11 by supermerill