supermerill/SuperSlicer #2642: Auto infill speed seems slow?

articnova, 2022-04-05:
### What happened?

Setting auto sparse infill speed seems to be showing as a slower than expected speed. Is there a setting that is causing this?

Setting sparse infill speed to a larger number manually also does not change the estimated speed so not sure which setting is preventing higher speeds.

Version 2.3.57.8

[image]

[image]

### Project file & How to reproduce

[z_drive_main_a_x2.zip](https://github.com/supermerill/SuperSlicer/files/8416944/z_drive_main_a_x2.zip)

### Version

2.3.57.8

### Operating system

Windows 10

neophyl, 2022-04-05:
Change your preview to Volumetric Flow.  You will see that the infill is being limited to the 11mm/3/s that you have your Max Volumetric Flow configured to under your Filament profile.

Because your infill extrusion id wider (180%) its laying down more plastic so while the speed is less its still putting out more in the same time.

articnova, 2022-04-05:
@neophyl  Ahhh thank you, I totally forgot about the filament specific volumetric flow. Altering that has fixed my query so I will now close the issue.

closed 2022-04-05 by articnova