supermerill/SuperSlicer #2644: G-code substitutions run after Print remaining times

Lecso11, 2022-04-05:
### What happened?

I wanted to replace the _M117 Time Left 0h3m53s_ line with a line such as _MY_MACRO R=0h3m53s_ so I could pass the time as a variable, but the Gcode substitutions might run before SS adds the M117 lines, and that results in no change in my code, since it can't find those lines.

### Project file & How to reproduce

[OpenLOCK-Clip-v5.3.zip](https://github.com/supermerill/SuperSlicer/files/8421408/OpenLOCK-Clip-v5.3.zip)
Enable Print remaining times with M117
Enter a Gcode substitution for _M117 Time Left_ and replace it with _R=_
The resulting gcode does not have the M117 lines changed.

### Version

2.4.58.2

### Operating system

Windows 10

### Printer model

Voron 2.4 with klipper

neophyl, 2022-04-06:
Its my understanding that the M117 remaining times are produced by a post processor, that means it has to have the gcode done before it can produce them.

I dont know how feasible it is to control the ordering of any /all of these various post processing options that we seem to have now though.  Have to wait for Merill on that one.

supermerill, 2022-04-07:
I guess what you have to do for now is to create a post-process script that replace `M117 (.*)` by `MY_MACRO R=$1`

I won't change the behavior of the gcode replace current alog in the near future, as I already ahve a bit too much thing to change.

aleung, 2022-05-07:
A workaround is to call `sed` in post processing script. (It's available on Linux. Need to find alternative on Windows) 

I use this post-processing script to turn off bed heating 4 minutes before printing finish:

```
/usr/local/bin/sed -i 's/M117 Time Left 0h4m0s/M140 S0/';
```