supermerill/SuperSlicer #2645: Modified custom gcode doesn't trigger "changed" flag, Save remains  disabled

CuredPrusa, 2022-04-05:
### Version
Version: 2.4.58.2
Build: SuperSlicer_2.4.58.2
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### Behavior
Modified custom gcode doesn't trigger "changed" flag, Save command remains disabled (grey)

supermerill, 2022-04-06:
what custom gcode?
"changed" flag = remove curent preview?
"Save command remains disabled (grey)" what is that save command?

CuredPrusa, 2022-04-06:
Save project CTRL+S remains grayed out even if I change something in the additional custom gcode in the ruler on the right. 
I ususally hit CTRL+S in my workflow, but in this case I have to go to "Save As" instead.
I guess a flag singnaling that something was changed ("changed" flag) in not triggered by editing a custom gcode.