supermerill/SuperSlicer #2647: Fractal Pyramid Vase Mode slice fail.

ssna, 2022-04-06:
### Version
2.4.58.2

### Operating system type + version
Windows 10

### Behavior
Does not slice "Fractal Pyramid" correctly in Vase Mode.  It is possible to do this in Vase mode in Cura.  It's very interesting that you e Vase Mode can wok with a model like this.

Not worried about this model but it may point to a fix that is worthwhile.

Slice in Vase Mode is missing a lot of fractal sections.
[image]
[image]

_Is this a new feature request?_
No.  This is an unusual file being Fractal making Vase mode quite complex.  Might be a simple logic mis-step otherwise it might be very difficult to track down.

#### Project File (.3MF) where problem occurs
[pyramid4.zip](https://github.com/supermerill/SuperSlicer/files/8424991/pyramid4.zip)

Gcode sample
[pyramid4 Generic PLA lw 0.4mm lh 0.2mm if 0% ext1 200C bed 50C.zip](https://github.com/supermerill/SuperSlicer/files/8425055/pyramid4.Generic.PLA.lw.0.4mm.lh.0.2mm.if.0.ext1.200C.bed.50C.zip)

_

neophyl, 2022-04-06:
If you scale up the model to 200% with your settings you will see it slices fine.
This would suggest that its something in your settings causing the issue.  Probably your extrusion widths are too large for the model size.
Looking at it further you have wall thickness set to 0.8, if I change that single parameter to 0 with the model scaled as is,  then it also appears to slice fine.  

You also have settings such as vertical holes shrinkage values set which can also change how a object slices.  Basically there are a million combinations of things that could effect a slice.  Best rule them out first so thanks for including a project file so we can see the settings being used.

ssna, 2022-04-11:
Thanks for your feedback

closed 2022-04-11 by ssna