supermerill/SuperSlicer #2649: Width and spacing being improperly calculated

Sirnut, 2022-04-06:
### What happened?

In the latest 2.4 release I have been playing around with the extrusion width and spacing and have noticed an odd issue

For the first layer width, if I set width to anything but the default 141%, the spacing goes to 100%. If I adjust the spacing, width appears to be calculated appropriately

Again focusing on **First Layer** values, the default width & spacing are 141% and 130% respectively
Changing width to 140% switches spacing to 100%
Changing spacing to 100% switches width to 111%

### Project file & How to reproduce

[width and spacing.zip](https://github.com/supermerill/SuperSlicer/files/8428306/width.and.spacing.zip)
t

### Version

2.4.58.2

supermerill, 2022-04-07:
seems fixed in the last nightly

xenon2008, 2022-04-24:
same error i found in the 2.3.57.12

closed 2024-02-11 by supermerill