supermerill/SuperSlicer #2650: Wipe tower last wipe in mid air

nikolai-matijevic, 2022-04-06:
### What happened?

When using the wipe tower with the `No sparse layers (EXPERIMENTAL)` option, the last wipe is placed in mid air instead of on top of the wipe tower. This happens on 2.3.57.11 as well as 2.4.58.2. 

[image]

I would expect the last wipe to be on top of the wipe tower which is on a lower layer in this case.

### Project file & How to reproduce

[Support Test v2.zip](https://github.com/supermerill/SuperSlicer/files/8428564/Support.Test.v2.zip)

Be sure to enable `Wipe tower` and `No sparse layers (EXPERIMENTAL)` under `Print Settings` -> `Multiple extruders`

### Version

2.4.58.2

### Operating system

Windows 10

neophyl, 2022-04-06:
https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)  This is from the Prusa slicer faq as this was raised as an issue on their GitHub. Second item down. So not a bug apparently.

nikolai-matijevic, 2022-04-06:
Thank you. I'll close it then.

closed 2022-04-06 by nikolai-matijevic