supermerill/SuperSlicer #2651: Parallel lines on top of bridge overhang

JiangMinyang, 2022-04-06:
### What happened?

The layer on top of over hang is parallel to over hang lines, not perpendicular. This causes layer adhesion problem with the overhang layer.[](url
[image]
[image]
)

### Project file & How to reproduce

Turn the model 45 degree (main axis on diagonal of the bed). Slice this stl with support turned on, 0.2 layer height, 0.24 on first layer.  The issue appears on layer 80/81.
[model.zip](https://github.com/supermerill/SuperSlicer/files/8429867/model.zip)

### Version

2.4.58.2

neophyl, 2022-04-07:
You could just set the bridging angle to something else.  Thats what I normally do in these cases.

supermerill, 2022-04-07:
Bridge lines are computed automatically to be on the best orientation.
Infill angle are taken from the setting, and rotated by 90° each layer (+the increment setting).

I don't see how to resolve your problem without rotating the part / changing the settings.

supermerill, 2022-04-07:
> Why + increment?

because it was requested, and there is some uses to it, like cancelling the 90° change at each layer.

> Or just specify repeating pattern of angles like it is done in the simplify3d

duplicate of #2557

closed 2022-04-07 by JiangMinyang