supermerill/SuperSlicer #2652: Adaptive layer height not working

StarGate01, 2022-04-07:
### What happened?

My installation does not manage to actually modify the heights of the generated layers when I enable adaptive layer height.

What am I doing wrong? Thank you for any help.

### Project file & How to reproduce

Project file: [Hemisphere_test.zip](https://github.com/supermerill/SuperSlicer/files/8439423/Hemisphere_test.zip)

Images:
[image]
[image]
[image]

### Version

2.3.57.12

### Operating system

Windows 10 x64

### Printer model

Anet AM8 with SKR2 and BIQU H2

qwewer0, 2022-04-07:
Works fine on 2.4.58.2
[image]

StarGate01, 2022-04-07:
Oh sorry, I did not realize there was a pre-release branch. Now it works fine for me as well, thank you!

closed 2022-04-07 by StarGate01

supermerill, 2022-04-07:
Seems like in the 2.3 branch, stl that are auto-repaired can't have variable layer height, or something like that.