supermerill/SuperSlicer #2654: Infill bug in vase mode

btthrt, 2022-04-07:
### What happened?

No infill in these areas
[image]
s

### Project file & How to reproduce

Slice this file in vase mode with 6 bottom layers and whatever infill pattern you want. 
Get almost no infill in tight areas, only bottom and top layers hanging
[VaseV649.zip](https://github.com/supermerill/SuperSlicer/files/8441978/VaseV649.zip)
.

### Version

2.3.57.12

### Operating system

windows 11

### Printer model

voron 2.4

btthrt, 2022-04-08:
Ok, small clarification, set external perimeter to something close to 0.6-0.7mm for .4 nozzle
Then try slicing

btthrt, 2022-04-09:
@supermerill sorry, I'm a little bit retarded

Here, please verify my stupidity. :)

[VaseV649 bugged.zip](https://github.com/supermerill/SuperSlicer/files/8457305/VaseV649.bugged.zip)

supermerill, 2022-04-11:
That comes from a code that prevent too much solid infill when sparse infill is set at 0. Will deactivate it for spiral vase.
comment from code, for reference:
```
                            // if we're printing a hollow object we discard any solid shell thinner
                            // than a perimeter width, since it's probably just crossing a sloping wall
                            // and it's not wanted in a hollow print even if it would make sense when
                            // obeying the solid shell count option strictly (DWIM!)
```