supermerill/SuperSlicer #2655: Modifiers don't work for fuzzy skin

wnc2, 2022-04-07:
### What happened?

i add a modifier to avoid fuzzy skin in a certain area, but no changes. Prusaslicer does it correctly

[image]

[image]

### Project file & How to reproduce

[fuzzy.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8442030/fuzzy.3mf.zip)

### Version

2.4.58.2

### Operating system

macos15

### Printer model

ratrig V-core 3

wnc2, 2022-04-07:
it happens on 2.4.58.2, on 2.4.58.3 the same job crashes the program

supermerill, 2022-04-07:
Can't reproduce on windows  (using the latest nightly)
[image]

supermerill, 2022-04-07:
just to be sure, as there was  a bug with modifiers: did you try by loading your project after saving it?

wnc2, 2022-04-07:
The latest 58.3 for macOS crashes on the same job. If you want I can send you Apple report

liftbag, 2022-04-07:
Strange, but it seems a macos related issue. I can reproduce the issue on macos 12.3.1. It crashes (segmentation fault: 11) as soon as you add a fuzzy skin modifier (or height range).

liftbag, 2022-04-07:
> just to be sure, as there was a bug with modifiers: did you try by loading your project after saving it?

New project, add a shape box, add height range modifier, add Fuzzy skin to the modifier.

wnc2, 2022-04-07:
I have the project file for you, To send you my settings,I use a simple STL file, I load it every time I need it

wnc2, 2022-04-07:
Beside I try to use modifiers, because fuzzy skin works badly on the upper edges of that box, fuzzy are applied also on top and on the edges, also when “outer walls” is selected

wnc2, 2022-04-07:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [43336]
Path:
 /Users/USER/Downloads/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.3_2022-04-04
(SuperSlicer_2.4.58.3_2022-04-04)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-04-07 13:46:29.8231 +0200
OS Version:            macOS 12.3 (21E230)
Report Version:        12
Anonymous UUID:        87B578F4-F403-F16E-7F9B-AAC282E6B5E1

Sleep/Wake UUID:       F0F9EA8C-C0F3-49CE-A11B-A353DEC354CE

Time Awake Since Boot: 130000 seconds
Time Since Wake:       14534 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000048
Exception Codes:       0x0000000000000001, 0x0000000000000048
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [43336]

VM Region Info: 0x48 is not in any region.  Bytes before following region:
4558675896
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX
SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->
      __TEXT                      10fb7e000-111a66000    [ 30.9M] r-x/r-x
SM=COW  ...S/SuperSlicer

Application Specific Information:
Performing @selector(clickedAction:) from sender wxNSMenuItem 0x6000035d50a0

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   SuperSlicer                         0x1113a0464
wxSizer::GetItem(unsigned long) + 4
1   SuperSlicer                         0x110681ed8
std::__1::__function::__func<Slic3r::GUI::create_settings_popupmenu(wxMenu*,
bool, wxDataViewItem)::$_73,
std::__1::allocator<Slic3r::GUI::create_settings_popupmenu(wxMenu*, bool,
wxDataViewItem)::$_73>, void
(wxCommandEvent&)>::operator()(wxCommandEvent&) + 1800
2   SuperSlicer                         0x11157c99e
wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
3   SuperSlicer                         0x11157c70b
wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
4   SuperSlicer                         0x11157c620
wxEvtHandler::ProcessEvent(wxEvent&) + 96
5   SuperSlicer                         0x11157cb3c
wxEvtHandler::SafelyProcessEvent(wxEvent&) + 12
6   SuperSlicer                         0x1113845b2
wxMenuBase::DoProcessEvent(wxMenuBase*, wxEvent&, wxWindow*) + 130
7   SuperSlicer                         0x111384494
wxMenuBase::SendEvent(int, int) + 196
8   SuperSlicer                         0x11144c0b1
[... 729 lines cut ...]
USB Device: VLI Product String
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: UHS-II SD Reader
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.1 Hub
USB Device: USB3.1 Hub
USB Device: USB3.0 Hub
USB Device: Cam Link 4K
USB Device: USB2.0 Hub
USB Device: Yeti Stereo Microphone
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: Stream Deck
USB Device: USB2.0 Hub
USB Device: USB Receiver
Thunderbolt Bus: iMac, Apple Inc., 47.5

wnc2, 2022-04-07:
prusaslicer and superslicer compared on default fuzzy skin

[image]

wnc2, 2022-05-05:
problem seems solved with latest night build
[image]

supermerill, 2022-07-14:
Same, can't reproduce

closed 2024-02-11 by supermerill