supermerill/SuperSlicer #2656: page "zh_cn > FilamentSettings > Cooling" crash

BlackTeaLeaf, 2022-04-07:
### What happened?

The Chinese simplified "FilamentSettings > Cooling" page will cause the program to crash
[image]

Expect to display and operate normally

### Project file & How to reproduce

'FilamentSettings' -> 'Cooling'

### Version

2.3.57.11 and 2.3.57.12

### Operating system

windows 10

supermerill, 2022-04-07:
probably a setting placeholder missing in the translation. I'll look.

BlackTeaLeaf, 2022-05-06:
[filament.zip](https://github.com/supermerill/SuperSlicer/files/8638124/filament.zip)

I found it was a configuration problem

"PLA 透明 A20-1KG-0114-21-K1.ini" ok
"PLA 黑 A20-1KG-1225-20-K1.ini" crash

I used winmerge to compare and found no noticeable difference that might cause the crash
but the crash still occurs

[image]

supermerill, 2022-05-06:
I think it's because the text at the bottom is created from the different cooling settings, and so the formatting error only appear on a specific cooling combination

supermerill, 2022-05-06:
found it, it's the translation of `between %1%%% %2%%% over external perimeters`
currently set to `外围轮廓` but it needs to contain `%1%` and `%2%` somewhere (`%%` will write a `%`, that's why there is %% after the %X%).

BlackTeaLeaf, 2022-05-09:
Does this need to wait until release to verify?

supermerill, 2022-05-18:
> Does this need to wait until release to verify?

i already replaced it by "外围轮廓 %1%%% %2%%%" to stop crashing, but you should have a better translation?

BlackTeaLeaf, 2022-06-08:
simplified chinese is not my native language
I may not be able to give ideal advice
I think just need to correct the translation parameter