supermerill/SuperSlicer #2658: No temperature control tick box

mirasderbissalin, 2022-04-07:
I want to control temperatures on a printer by printer basis, (nozzle material, different build surfaces). Obvs, I can just create different filament profiles for printers, but that's a hassle, and tweaking them every time I change a manufacturer will be a logistical nightmare that is bound to produce mishaps. 

**Describe the solution you'd like**
A tick box that would filter out every temperature command in produced Gcode (in printer settings tab I imagine)

**Describe how it would work**
SS won't inject any M104, M109, M140, M190 commands if the box is ticked

**Describe alternatives you've considered**
I tried "Use only custom start Gcode" with fist layer temperature equal to the rest of the print, but it still injected temperature commands at second layers. Also, I had to put START_PRINT macro in filament settings, bc SS puts filament-specific Gcode after start gcode, so heats up after homing and priming, is there is a way to swap that order? If not, would be nice to have an option

**Additional context**
Not super crucial, suited for someone who has drastically different printers

neophyl, 2022-04-08:
Prusa Slicer recently added their built in gcode substitution/processor, this is in the 2.4.58 branch of SuSi as its built on PS 2.4.x.  This should allow you to do what you want already.  Its not under Printer Setting but Print Settings>Output Options.  Its a red setting though so you need to be in expert mode to see it.

mirasderbissalin, 2022-04-08:
Ok, waiting for stable 2..4.58 then, thanx) I am in expert mode, great