supermerill/SuperSlicer #2659: Bridging Angle

nicholasbaker, 2022-04-07:
V. 2.3.57
Windows 10
MKS3+

Been loving superslicer over prusaslicer, but one thing I noticed is different on the bridging angle.

In Prusaslicer when you put in the 180 vs 360 deg. bridge angle this will change the direction of printing over a hole. I prefer to print bridges parallel to the x axis starting in the back and moving to the front for maximum air cooling. 

In Superslicer 180 vs 360 seems to have no difference on how whether the bridges start in the back or front. I would like to have this option.

Cheers,
Nick

supermerill, 2022-04-08:
what do you mean by "start in the back or front" ?

nicholasbaker, 2022-04-08:
[image]

This is how Prusaslicer works when you type in those two angles. Superslicer is inconsistent with which way it will bridge, seems based on shortest travel distance.

supermerill, 2022-04-08:
It's just luck, comapre 90° and 270°