supermerill/SuperSlicer #2660: Gcode preview doesn't show multiple extruders if Tool change G-code section has content

HitLuca, 2022-04-07:
### What happened?

I recently noticed that my multicolor prints don't show as such in the Gcode preview when I slice them.
My personal setup involves a single extruder single material printer, for which I manually change filament when needed to obtain multi color prints. When slicing them, I notice that only `extruder 1` is shown in the Gcode preview tab

[image]

In order to debug this issue, I imported a Mk3S+ MMU2 printer, and resliced my file. Works as expected

[image]

I narrowed the problem down to the Printer `Settings` -> `Custom G-code` -> `Tool change G-code`: if anything is added in this section, the Gcode preview tab doesn't work as expected

Anything goes there, even an empty comment (`;`), breaks it.

### Project file & How to reproduce

- add a multi material printer profile to SuperSlicer
- make sure the  `Tool change G-code` section is empty
- slice a multicolor print, make sure multiple extruders are correctly shown in the Gcode preview
- add any command in the `Tool change G-code` section
- reslice the print
- now only `extruder 1` is shown in the preview tab, with only one color

### Version

2.3.57.12

### Operating system

Ubuntu 20.04

### Printer model

Eryone Thinker SE, and MK3S+ MMU2 for testing

supermerill, 2022-04-08:
from the `toolchange_gcode` tooltip:

```
This custom code is inserted at every extruder change. If you don't leave this empty, you are 
expected to take care of the toolchange yourself - Slic3r will not output any other G-code to 
change the filament. You can use placeholder variables for all Slic3r settings as well as {toolchange_z}, {layer_z}, {layer_num}, {max_layer_z}, {previous_extruder} 
and {next_extruder}, so e.g. the standard toolchange command can be scripted as T{next_extruder}.
!! Warning !!: if any character is written here, Slic3r won't output any toochange command by itself.
```

=> add the toolchange command yourself:  `T{next_extruder}`

HitLuca, 2022-04-08:
Thank you very much for the quick response!
I would really like to have the Gcode preview tab working as I would like, so I will add the toolchange command as you suggest and see if I can filter it out on my octoprint server

closed 2022-04-08 by HitLuca

BenLand100, 2023-07-30:
Encountered this today... my 2 cents: while the tooltip does clarify the default, and that the default won't be emitted if anything is added (this, I think, is expected behavior), it doesn't clarify that omitting the default tool change commands (an M600 is all I need) will result in no tool changes being detected by the slicer visualization _AND_ no generation of custom tool change gcode (M600) in the export!

To clarify, a project with this custom tool change gcode results in a `T0 is not a valid command` (firmware does not support tool changes) but does do an M600 at the right place:
```
T{next_extruder}
M600
```
This custom tool change gcode never emits an M600, even if tool changes are requested. Visualization also shows no tool changes, even if requested.
```
M600
```
 This, i think, is a bug. I'd expect tool changes to emit this M600 at the correct layer changes.