supermerill/SuperSlicer #2661: brim and support don't play well together. plz fix

maz323f, 2022-04-07:
### Version
Version 2.3.57

### Operating system type + version
win11

### Behavior
Brim and supports dont work together..

the brim go on the supports and not the part..  plz fix that so it works like the other slicers do.  they lay the brim on the part not the support..

*attatched file snows superslicer to the left. cura in the middle and prusa to the right..

[image]

Ugars, 2022-05-09:
Using 2.4.58.2 and having the same problem. If you enable support and brim somehow brim moves next to support edge and becomes unfunctional, because now there is a gap between support bottom and edge of the model. 
[Support_Brim_PROBLEM_0.2mm.zip](https://github.com/supermerill/SuperSlicer/files/8654864/Support_Brim_PROBLEM_0.2mm.zip)

Please check attached 3mf and gcode files to see the problem.

AssiveMass, 2023-03-29:
This is a big issue that's been fixed in Prusaslicer like two years ago, Please implement it here too :)
https://github.com/prusa3d/PrusaSlicer/commit/055d2321e7d0b5727e1933a8c64ef6074bc2956c