supermerill/SuperSlicer #2664: SuperSlicer Locks up, CPU goes to 100%

notching, 2022-04-08:
### Version
2.3.57.11 

### Operating system type + version
Windows 10

### Behavior
Superslicer locks up when trying to slice, CPU goes to 100%.

[SpringFlat_Soft_v3.3.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8450317/SpringFlat_Soft_v3.3.3mf.zip)

supermerill, 2022-04-08:
It's not really stuck, it just takes a very very long time.
The problem lie in the shape & the algo for gapfill. The shape makes hte current algo to be on its worst-case scenario, and the spiral creates a very high amount of points/paths.

One simple solution is to revert the resolution field to its default value of 0.0125 to prevent the humongous amount of points to stall the slicing.

notching, 2022-04-08:
Thanks for the tip on the resolution, appreciated, I can slice now :-).

Cancel did not work for me so I assumed it was stuck.

supermerill, 2022-04-08:
The cancel is check before & after that step, as it should be computed quickly.

supermerill, 2022-04-08:
I added some failsafe to decrease the resolution if the gapfill is detected with too may points.

What a huge gapfill btw.
[image]

closed 2024-02-11 by supermerill