supermerill/SuperSlicer #2665: Infill - Significant overextrusion for small areas

TommyC81, 2022-04-08:
### What happened?

Infill of small areas leads to excessive over extrusion without any clear option to fix it without affecting larger infill areas negatively.

Options exists to reduce general "Filament Settings" -> "Max line overlap", and there are also options "Print Settings" -> "Width & Flow" -> "Solid fill overlap". However, reducing those numbers will impact larger infill areas where the issue may not be prevalent.

In the image below (and attached zip file), I've specifically setup a test piece to show examples of small infill areas. I have reduced the "Solid fill overlap" setting to 25% but am 
[image]

From what I understand, this is likely an artifact of anchoring infill (which is required, but will cause some over extrusion itself), along with general line overlap that simply cannot be handled appropriately in small infill areas.

It seems the solution would be a flow modifier (or equivalent) for infill areas that are smaller (or narrower) than a certain size to avoid this, without impacting on larger infill areas. Perhaps scaling with the size/shape of the infill area.

Example project:
[Testpiece.zip](https://github.com/supermerill/SuperSlicer/files/8451344/Testpiece.zip)

The issue is prevalent with gap fill as well, but the "Gap Fill" overlap setting can (largely/completely) fix this without affecting any other area (as it is just that - small pieces of gap fill, it's own type of line used in only small spaces).

### Project file & How to reproduce

Open attached project file, print, check for over extrusion in areas circled in red.

Note: "Gap Fill" overlap, and "Solid fill overlap" are both reduced to 25% in the project file, this reduces the issue but will affect larger infill areas (signs of minor under extrusion). Change the settings to 100% to print with excessive over extrusion = mountains of plastic twice the height of the layer.

### Version

SuperSlicer nightly_dev 2022-04-06 (bfb10586dc0ea7a4e2ef4767be6c181043495410)

### Operating system

Windows 10

### Printer model

Creality CR6 with Sherpa Mini direct drive

supermerill, 2022-04-08:
With 100% overlap and "enforce 100% fill volume", the volume extruded is exactly the empty volume dedicated to the infill, whatever the "infill/perimeter encroachment" is.

TommyC81, 2022-04-08:
Thank you, the option is selected.

The help-text mentions that "...it can go as high as +50% for infill in very small areas where rectilinear doesn't have good coverage)...". 

Could that be what is being triggered to cause the excessive over extrusion in the red circled areas?

supermerill, 2022-04-08:
Unlikely. to trigger more than a 5% increase, the area has to be very small, like an area where two lines can't be printed, but almost.

Your issue is probably a too high filament extrusion multiplier.

TommyC81, 2022-04-08:
E-value is tuned on the printer, extrusion multiplier is 1. Getting perfect output except in the small infill (and Gap Fill areas).

With "Solid infill overlap" at 100% I get significant over extrusion in the small infill areas (probably 3 times the layer height of blobs sticking up, lots of loud nozzle bumping as a result). With "Solid infill overlap" reduced to 25% I get about half of that, but start to see (light) under extrusion signs in larger infill.

supermerill, 2022-04-08:
small over extrusion on small areas vs big ones is something that is a bit hard to resolve.
Often, the problem raise when a big area has a thin section, and that section will get more plastic (from the encroachment) and so will appear a bit over-extruder vs the rest f the area.

But here, you have only almost-square area, that should distribute the plastic pretty evenly.

Maybe the issue comes for the pattern that can't distribute the plastic evenly enough fo it to flow everywhere.

TommyC81, 2022-04-08:
Thanks! Yeah, I gather it can get complicated as the area sizes can be mixed and run into (very) different considerations for the same settings.

I'll try with those settings and let you know.

To check; changing ironing distribution and width setting will also affect the top infill then (I gather)?

supermerill, 2022-04-08:
To check; changing ironing distribution and width setting will also affect the top infill then (I gather)?

Only for the ironing pattern (don't have an effect on the ironing post-process as it's a separate feature)

TommyC81, 2022-04-08:
Thank you!

So, I did a quick run with the 30% distribution and 100% width (infill: ironing). I also re-set "Solid infill overlap" to 100% to do it fresh this run.

Already at the first 70% pass there were obvious over extrusion in the infill. The 30% ironing cleaned it up somewhat, but mostly only moved the bumps of plastic around the top - the outcome per layer was however much cleaner than a pure 100% infill, without 30/70 distribution with an ironing pass, but still with bumps about twice the layer height left over (visual inspection). 

Another minor artifact I noticed is that the over extrusion seems to have pushed the external perimeter (the south-eastern wall in my previous picture) out a bit - it's slightly curved outwards instead of straight. I assume this is because over extrusion on lower layers, causing upwards bulges, has caused the upper perimeters to flow outwards (as there was not the expected internal space available for the perimeters to flow inwards as well).

TommyC81, 2022-04-08:
> > To check; changing ironing distribution and width setting will also affect the top infill then (I gather)?
> 
> Only for the ironing pattern (don't have an effect on the ironing post-process as it's a separate feature)

Just to confirm, these were the ironing settings:

[image]

supermerill, 2022-04-08:
yes.

Note that I disregard any top setting, as it's not the subject. We are testing the solid infill.

TommyC81, 2022-04-08:
I will try another piece, with a slightly adjusted test piece to see what comes out of it.

Note that I've reverted all settings to 100% infill/gap fill again, just to compare the smaller vs the (relatively) larger infill areas.

[image]
[Testpiece2.zip](https://github.com/supermerill/SuperSlicer/files/8454094/Testpiece2.zip)

TommyC81, 2022-04-08:
Outcome:

First layer looks great, smooth without any sign of over extrusion during the print. In the completed print I can just about make out the printed lines, so I guess I could up the flow rate slightly (so perhaps the top is also ever so slightly under extruded as well, but can't really see it as I checked during the print).

Beyond the first layer:
Black circle: Obvious over extrusion from layer 2-3 onward (blobs apx. 1-2 layer thick, occasional nozzle bump during travel. 
Red circle: Significant over extrusion from layer 2-3 onward (blobs apx. 2-3 layer thick, loud nozzle bumping as it passes during travel).
Blue circle: Slowly accumulating over extrusion. As it reaches the last layers, there's a few blobs sticking up about 1 layer height above the print.

[image]

Separate observation:
The thin wall is printed in two passes, first with an incredibly thin line (0.08mm) at the start of the layer, and later with a over-printed with a "normal" line:
[image]
[image]

TommyC81, 2022-04-08:
Tweaked settings a bit:

First layer flow 102% - as previous (updated post), slight sign over extrusion (blob) as it finishes the print in the black and red circle, and the lines on the bottom are almost gone.

Gap Fill and Solid infill overlap both reduced to 50%, this reduced overall infill over extrusion to:
Black circle: Visible over extrusion from layer 2-3 onward (blobs apx. 1 layer thick, occasional nozzle bump during travel.
Red circle: Marked over extrusion from layer 2-3 onward (blobs apx. 1-2 layer thick, occasional nozzle bump during travel.
Blue circle: Slowly accumulating over extrusion. As it reaches the last layers, there's a few blobs sticking up about 1 layer height above the print.

I hope this helps, let me know if there's anything specific you want me to test.

[Testpiece3.zip](https://github.com/supermerill/SuperSlicer/files/8454359/Testpiece3.zip)

supermerill, 2022-04-09:
> The thin wall is printed in two passes, first with an incredibly thin line (0.08mm) at the start of the layer, and later over-printed with a "normal" line

should be the opposite.

TommyC81, 2022-04-09:
> > The thin wall is printed in two passes, first with an incredibly thin line (0.08mm) at the start of the layer, and later over-printed with a "normal" line
> 
> should be the opposite.

Yeah, I think the initial super-thin line shouldn't happen at all. It doesn't cause any issues however (at least not in this test file). The "normal" line at the end of the layer is all that should be printed.

For reference, it can be seen in any of the previously attached project files.

TommyC81, 2022-04-10:
Another example with significant over extrusion in small infill areas (even with Gap Fill/Solid fill overlap at 25%, and overall filament Max line overlap turned down to 80%):

The over extrusion is causing major blobs in the print (2-3 layers high), it's more or less a barrier around the nozzle as it prints - the plastic is bulging up around the nozzle. And on subsequent movements/layers, there's extensive nozzle hits over these small infill areas. On larger infill areas, the infill is more or less perfect.

[image]

[image]

[3DBenchy.zip](https://github.com/supermerill/SuperSlicer/files/8458751/3DBenchy.zip)

TommyC81, 2022-04-10:
A further follow-up with pictures. Can see in Layer 2 already that things are starting to go off with the bridge infill, then there's further over extrusion in layer 3. And so it continues for all small infill areas. I've re-calibrated e-steps again to check that all is set correct (it was).

[image]

[image]

[3DBenchy - 3.zip](https://github.com/supermerill/SuperSlicer/files/8458944/3DBenchy.-.3.zip)

TommyC81, 2022-04-10:
Did another Benchy - resetting all values to 100% infill etc. but lowering extrusion multiplier to 0.9. Got a good result (no over extrusion - including the previously observed extreme over extrusion at the first layers), but slightly under extruded. Will tune a bit further to see if I can find a balance point.

[3DBenchy - 4.zip](https://github.com/supermerill/SuperSlicer/files/8459585/3DBenchy.-.4.zip)

TommyC81, 2022-04-11:
And further follow-up; filament extrusion multiplier at 0.92, first layer at 106% (should be effective ~97.5%) and top layer at 105% (96.6% effective). Getting decent prints now avoiding the over extrusion on the infill layers, but hard to determine whether it is under extruding instead. Testing further.

[Testpiece6.zip](https://github.com/supermerill/SuperSlicer/files/8465944/Testpiece6.zip)

supermerill, 2022-04-11:
> but hard to determine whether it is under extruding instead

with same speed, weight down your big benchy on a precision scale, and compare with the slicer weight output (include the skirt & prime lines on the scale)

TommyC81, 2022-04-11:
> > but hard to determine whether it is under extruding instead
> 
> with same speed, weight down your big benchy on a precision scale, and compare with the slicer weight output (include the skirt & prime lines on the scale)

Thanks, hadn't considered that. Will give it a try as soon as time permits.

Meanwhile, I did test further by adding a square piece along the "testpiece" I previously posted (check the file name). Outcome was a nicely extruded "testpiece" (many small infill and gap fill areas), but the larger square piece was instead obviously under extruded with gaps between the lines at the top layer (and guesstimated under extruded somewhat in the non-visible infill layers). So, I'm still struggling with finding a setting that allows both larger and smaller infill areas to be appropriately extruded.

TommyC81, 2022-04-12:
A more detailed set of test pieces to find the issue. First layer is very slightly over extruded, nothing alarming. The print coming out after 5 layers look good, although there is some over extrusion on the way (in particular in the circled areas in the image below for the 10 layer print). From layer 5 onward there is marked over extrusion in the circled areas for the 10 layer prints, it's more or less almost an extra layer's worth of over extrusion that the final ironing scrapes off eventually - but there's obviously some noticeable nozzle bumping going on during the print.

I will test further as time permits, something to check prior:

Where does "Enforce 100% fill volume" work it's magic? If I set reduced overlap for solid infill/gap fill, will that be nullified by "Enforce 100% infill volume"? (I gather that reducing gap fill overlap is a strong candidate for the next test)

Is it correct that filament extrusion multiplier is applied last as a general modified of whatever all extrusion calculations has yielded?

I appreciate any other ideas that can resolve this. To note that I'm using a 0.6mm nozzle with a direct drive (e-steps 698, with repeated calibration yielding results between 690-710 - so this should not be the issue, filament max volumetric flow is additionally limited to a "safe" 10mm3/s). The closest similar issue I could find was #2348, although there are some other ones with similarities.

[image]
[Testpiece10.zip](https://github.com/supermerill/SuperSlicer/files/8474670/Testpiece10.zip)

TommyC81, 2022-04-12:
Had time to do a quick test with gap fill overlap set to 0%.  No difference.

For my understanding, the infill/perimeter anchoring will by default lead to some over extrusion as the overlapping extrusion is not compensated for, correct? Is it possible that it is simply this anchoring that causes over extrusion in places where the infill lines are short and %-wise has a larger overall perimeter overlap (i.e. possible over extrusion) per infill line?

I'm just speculating here to make a plan for my next test.

supermerill, 2022-04-13:
> Where does "Enforce 100% fill volume" work it's magic? 

just after the infill path computation, it compare the extrusion volume with the infill volume and apply a extrusion multiplier uniformly to the extrusions.

> If I set reduced overlap for solid infill/gap fill, will that be nullified by "Enforce 100% infill volume"? (I gather that reducing gap fill overlap is a strong candidate for the next test)

yes and no, sending of that you mean. It will create different extrusions, as they won't be as packed as before. The "Enforce 100% infill volume" will increase the flow to compensate, but the infill area is still reduced. I guess I have to make the "Enforce 100% infill volume" aware of the overlap ratio.

> Is it correct that filament extrusion multiplier is applied last as a general modified of whatever all extrusion calculations has yielded?

yes, it's applied when the gcode line is written. It's multiplicative anyway, so whatever the moment you apply it doesn't change the result.

TommyC81, 2022-04-13:
> Had time to do a quick test with gap fill overlap set to 0%. No difference.
>
> For my understanding, the infill/perimeter anchoring will by default lead to some over extrusion as the overlapping [...]

I was actually thinking about the "encroachment" setting. But am curious about the anchoring as well - will "encroachment" and "anchoring" respectively be a possible cause of over extrusion (or is this compensated for by "enforce 100% infill volume").

supermerill, 2022-04-13:
Anchoring (the infill one, right?) is just a way to decide where to switch from solid to sparse, so it's out of scope.

encroachment is compensated by "enforce 100% infill volume".

TommyC81, 2022-04-13:
> yes and no, sending of that you mean. It will create different extrusions, as they won't be as packed as before. The "Enforce 100% infill volume" will increase the flow to compensate, but the infill [...]

Thank you, just re-read this. So, from this I gather that if overlap is not accounted for (specifically adjacent perimeters, and also gap fill), then there will by default be some over extrusion occurring. This would, if I understand it correctly, be more prominent in any area where the perimeter to area ratio is large, such as smaller/thinner infill areas. In fact, any "corner" area would be prone to some over extrusion. Conversely, the impact on larger infill areas (in particular round shapes, where the ratio between perimeter to area is the lowest possible) would be less affected, if at all (visibly).

If I understood this correctly, this would correspond closely to my observations. And it would also mean that there is no setting to work around it with "enforce 100% infill volume" selected. I think the setting itself is great, but it could use some tweaking to account for the overlap (alternatively/additionally to set limits to max decrease/increase in extrusion, or something else that is useful?).

TommyC81, 2022-04-14:
Continued testing - no solution just yet.
[Testpiece15.zip](https://github.com/supermerill/SuperSlicer/files/8490232/Testpiece15.zip)

veehexx, 2022-04-22:
I'm seeing the same thing. granted i've not ran though the various sugestions in here to see if it improves things for me.

i'm new to superslicer so only been using it for a month or two; came from Cura. Still fairly new to 3d printing too (6-9months now )and learning plenty.

Basic stuff has been done; E-steps are bang on (both extruder manuf recommendation, and my own measurements), using SS's out-the-box default profiles for the Ender3 v2+0.2 quality Creality PLA, first layer is pretty nice but the solid infill areas are overextruding.

I did get some improvement by dropping flow to 50-60%, but then i get underextrusion in other areas; as you'd expect!

veehexx, 2022-04-22:
not when i posted, no. I've ran another print with Rectilinear (filled) & 'not connected' in Infill > Solid.

left is the print i did before posting (default SS i mentioned in my last post), right is the print with your sugestion. You can see the nozzle drag and it's practically identical with the overextrusion outcome.
the lower part with the rpi logo is 5 layers high.

https://imgur.com/xAyUjx4

right side print is 100% completed, the final layer feels very smooth, and ironing would make it perfect. Theres no overextrusion on these layers

veehexx, 2022-04-22:
I've a good print done, but involved setting flow rate manually on layer 2-4 to 50%. I did this via octoprint for the easiest on-the-fly change, and changing back to 100% once those layers were extruded.

layer1 is obviously the first adhesion layer, layer 5 is the top solid infill for my model.
i suspect i see issues on layer =<5 because it's full model size and no where for the excess to go.
layers >5 are partial area so the excess doesnt build up as much, and also have 15% infill for somewhere to escape to without showing artifacts on the print.

Definately saw underextrusion on first layer if i set flow to 50% for the entire print and no/poor adhesion.
i think i saw underextruded top layers if i set 50% after the first. Both of these reinforces my thought on having a correctly calibrated extruder/flow rate (again, i'm still fairly new to 3d printing!)

i'm running 2.3.57. I'm yet the get the latest stable or try the dev version... I've been going down the one change at a time methodology to trace where the issue could be.

veehexx, 2022-04-23:
I've updated the the latest stable version now; 2.3.57.12.
something that came to mind last night was z-step calibration. Looks like theres something funky going on and i'm not sure where.
i'm using https://www.printables.com/model/84138-pi-zero-2-slim-case/files 02.stl model.

SS is showing 3.40mm, 17 layers.
printed version is measuring 3.04mm on either an overextruded print or my own 50% flow rate hacky fix. I'll look into this today; possibly a z-step calibration issue there.

While this model doesnt require support, if you enable support then it changes to 3.40mm, 20layers. where have these extra 3 layers come from? The SS previews still remains flat on the bed and theres no erroious support that is adding an extra 3 layers.
I guess what i'm getting at, is SS correctly handling layers? just as an example: if the layer height is 0.2mm, but flow rate is being miscalculated for a 0.3mm layer then it's obviously going to be using too much filament.

TommyC81, 2022-04-23:
I'm following this closely, although I've not had time to continue with further testing. Anyhow, to give a quick update. The attached is a project file from the latest nightly_dev (c467986ba991519465f3608e8520d4fff8ef41a3). I'm still unable to get it to avoid over extrusion in the "smaller" solid infill areas.

Interesting take that it might be extruding based on thicker layers than what is the case. After the tweaking and testing I've made I generally see more or less a layer worth of over extrusion in the small infill areas (where it is most evident, it may well be somewhat over extruded in larger infill areas as well).

[Testpiece16.zip](https://github.com/supermerill/SuperSlicer/files/8546985/Testpiece16.zip)

TommyC81, 2022-05-16:
An update as I'm still testing to see if I can resolve this via settings. So far the answer is no, unfortunately.

I've reduced line width and it works a little better, but there is some kind of bug causing silly amounts of over extrusion, more or less two layers worth of material is extruded at times.

The reduced layer width (along with reduced extrusion multiplier etc.) has kept the first layers _mostly_ in check (sort of).

Towards the top layer, there is significant over extrusion. 

[image]

About a layer worth scraped off by the ironing. (Also note a hole in the top of the dome).

[image]

Project file:
[Testpiece22.zip](https://github.com/supermerill/SuperSlicer/files/8697960/Testpiece22.zip)

Version used: nightly_dev (2e11c485266cf22987ae0f7be395df6a1295148c)

TommyC81, 2022-05-16:
@supermerill I've done some more extensive testing here, tagging you for awareness.

Version: nightly_dev (2e11c485266cf22987ae0f7be395df6a1295148c)

Test pattern:
[image]

Test pattern STL: [TestPrint1 - Infill.zip](https://github.com/supermerill/SuperSlicer/files/8701967/TestPrint1.-.Infill.zip)

Project file: [Testpiece23.zip](https://github.com/supermerill/SuperSlicer/files/8701984/Testpiece23.zip)

I stopped the print after the first layer.

I get small ridges in the first 5mm from the infill interaction with the perimeter. For longer infill (exceeding 2 * 5mm = 10mm), these minor ridges has no impact and the area in-between is very nice. However, for any infill distances shorter than 10mm, these ridges start overlapping making the ridges i.e. over extrusion significant - this get progressively worse the shorter the distance. Thus, small infill areas get significantly over extruded, while larger (longer infill lines, >= 10mm) infill comes out nicely.

Images for reference:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

To note that I'm using a Mini Sherpa direct drive and **0.6mm nozzle**, linear advance is used with K=0.04. I doubt this is related, but mentioning it for completeness. Using CR6 Community Firmware release 6.1 (https://github.com/CR6Community/Marlin/releases/tag/v2.0.8.1-cr6-community-release-6.1).

For comparison, CURA (5.0 BETA, latest) starts infill with one infill "perimeter" and does not *print* the short connecting line between infill lines, it just travels:
[image]

SuperSlicer *print* the short connecting lines between the infill lines, but does not do an infill "perimeter":
[image]

Raw gcode files: [Testpiece23 - gcode output (Cura 5.0 and SuperSlicer).zip](https://github.com/supermerill/SuperSlicer/files/8702077/Testpiece23.-.gcode.output.Cura.5.0.and.SuperSlicer.zip)

Could this make a difference for small areas?

TommyC81, 2022-05-18:
> > SuperSlicer print the short connecting lines between the infill lines, but does not do an infill "perimeter":
> 
> remove the connection then.

Ugh, yes, of course.... Set Solid/Top/Bottom to "Not connected", got the expected improvement (albeit some small holes along the inner perimeter). Pictures and project file below. Follow-up test print with smaller details still show some over extrusion, will test further and report back here.

[Testpiece24.zip](https://github.com/supermerill/SuperSlicer/files/8718185/Testpiece24.zip)

[image]
[image]
[image]
[image]
[image]

TommyC81, 2022-05-18:
Further testing, specifically looking at "Enforce 100% infill" and "Infill/perimeter encroachment".

Observations: "Enforce 100% infill" consistently leads to over extrusion. Without this option enabled, the result is much better. With a 0.6mm nozzle I seem to get best result using 0% "Infill/perimeter encroachment" (tested with 0% and 20% - 20% gave a noticeable over extrusion along the perimeter, which obviously also made smaller areas come out markedly over extruded).

Project file for testing: [Testpiece26.zip](https://github.com/supermerill/SuperSlicer/files/8718825/Testpiece26.zip)

veehexx, 2022-05-21:
i've a number of prints today. Initial impressions with the first small print appears good with "Enforce 100% infill" = off, and "Infill/perimeter encroachment" = 0%.
Is there a reason some experimental settings are on by default? surely by it's very definition things might not go to plan or cause problems so having 'Enforce 100% infill' = enabled seems wrong.

I haven't tried 'remove the connection', but may experiment with the prints depending how the above works out.

thanks for your findings @TommyC81 - for a novice like me, your findings are helpful!

TommyC81, 2022-05-23:
> i've a number of prints today. Initial impressions with the first small print appears good with "Enforce 100% infill" = off, and "Infill/perimeter encroachment" = 0%. Is there a reason some [...]

Good to hear it is useful, I've been figuring out the details for a while now...

To update you, these settings (among many others) seem to work best at the moment (I'm using a 0.6mm nozzle):

- Enforce 100% infill = off
- Infill anchoring (top/bottom/solid) = Not connected (this avoid the extra infill along the edges that cause over extrusion for me)
- Infill/perimeter encroachment = 20% (to avoid minor gaps in the infill along the perimeters)

However, and although I'm using 100% flow (without modification), I'm finding that the top layer gets under extruded, I'm thus using 105% flow for the top layer (with ironing).

veehexx, 2022-05-23:
> * Enforce 100% infill = off
>
> * Infill anchoring (top/bottom/solid) = Not connected (this avoid the extra infill along the edges that cause over extrusion for me)
>
> * Infill/perimeter encroachment = [...]

i've come to similar. I'm using the default 0.4mm nozzle on the Ender3v2.

* Enforce 100% infill = off
* Infill anchoring = connected, except Solid = not connected.
* Infill/perimeter encroachment = 10%

Ironing on. I found very slight underextrusion with encroachment = 0% (possibly 5% too).
no changes on flow and while my print is no longer with me and hard to get a good photo of black filament, the printout has been pretty flawless. By far the best i've ever had with the printer - no elephants foot, layers are equal reguardless of layer characteristics and no catching the nozzle on the previous layers overextrusions.

i've a bit more work to do dialing my machine in as while i'm at 0.95 extrusion multiplier in SS, i'm having to further adjust it down to 95% in octoprints 'Flow rate modifier' value but this is giving me reliable results after 4 printouts.

AndrewEllis93, 2022-11-08:
[image]

Top to bottom (worst to best)
(Top/Solid/Overlap)
C/C       40% overlap
NC/C    25% overlap
NC/NC 25% overlap
NC/NC 10% overlap

"Not connected" and lowered overlap appears to make a pretty significant difference. 
Additionally, forcing retractions on the top layer as documented [here](https://github.com/AndrewEllis93/Print-Tuning-Guide/blob/main/articles/troubleshooting/small_infill_areas_overextruded.md#the-magic-bullet-but-with-downsides) also helped a ton. 

Retractions: 
[image]

daterrell, 2022-11-08:
Been working with @AndrewEllis93 as well, I've also tested and settled on NC/NC 15% which helped a lot, the final magic was forced retractions on top layer 👍🏼

Fervin1, 2024-03-09:
Hi, Just wondering if a solution to the over-extrusion on the infill only was solved as I have the exact same issue and for the life of me I can't find a solution. thanks in advance :)