supermerill/SuperSlicer #2666: Z hop whilst travelling

mann1861, 2022-04-08:
**Is your feature request related to a problem? Please describe.**
In tuning tpu I've  struggled to reduce all issues due to oozing. If I have z hop on I get stringing, If I have z hop off I get blobs. This is also a problem with other filaments such as petg g that strings with z hop but without z hop it can sometimes collide with curled up parts leading to failure

**Describe the solution you'd like**
When crossing perimeters normal retract and wipe occurs. After leaving the part z hop occurs whilst travelling so the nozzle has lifted by the time it reaches any other parts. It then lowers like normal z hop once in position to continue extrusion
[image]

**Describe how it would work**
Checkbox under z hop for zhop to occur not within object. Height controlled by standard z hop parameter. Option for how far past object to start lift and option for lift speed = z hop speed or continuous over length

**Describe alternatives you've considered**
Standard tuning to optimise part quality to reduce stringing blobbing or curling

**Additional context**
Add any other context or paint drawing about the feature request here.

supermerill, 2022-04-08:
kind of duplicate of #927

mann1861, 2022-04-08:
Hadn't seen #927 whilst searching but yes similar idea however mine has the nozzle drop vertically once over the object instead of during travel as this will prevent blobbing. #927 solves the nozzle hitting the print but not blobbing

closed 2022-04-08 by mann1861

Lecso11, 2022-06-17:
Was this implemented?