supermerill/SuperSlicer #2667: Purge Tower Speed and Behavior

superblobmonster, 2022-04-08:
### What happened?

Purge tower speed inconsistent, speed controls give unexpected results.
This may be somewhere in the blurry line between bug and feature request, but I have not seen any clear guidance on expected controls for purge tower speed control. Perimeters and internal support of the purge tower doesn't seem to be affected by any speed settings. I don't discount the possibility that this problem exists somewhere between the keyboard and chair.
Worth noting, my settings are configured for the Mosaic Pallet 3, so no ramming settings, and no multi-material string reduction. Just one continuous feed of material.

Desired behavior: Purge tower should follow max auto-speed rules, and or have dedicated speed settings.

Autospeed volumetric flow rate set to 15mm^3/s, no settings in filament override
[image]
Same settings, two strange fast lines when filling in a new portion of the purge tower
[image]

Filament max speed on wipe tower set to 230%
[image]

Filament max speed on wipe tower set to 400% violates maximum print speed rules (300mm/s)
[image]
Those two strange lines are still the same speed:
[image]

### Project file & How to reproduce

[Export.zip](https://github.com/supermerill/SuperSlicer/files/8454496/Export.zip)

### Version

2.4.58.2

### Operating system

Linux Mint 20.1 Cinnamon, 5.13.0-39-generic

### Printer model

Voron 2.4

miguelangel-nubla, 2022-09-27:
I am also seeing this, ERCF on my side with ramming and all.

I have found the root of the issue, there is a 2900 feedrate hardcoded into the perimeter and carried over to the internal support [here](https://github.com/supermerill/SuperSlicer/blob/76856c584987547eb37ceb1140a891991b491dc0/src/libslic3r/GCode/WipeTower.cpp#L1404) from [here](https://github.com/supermerill/SuperSlicer/blob/76856c584987547eb37ceb1140a891991b491dc0/src/libslic3r/GCode/WipeTower.cpp#L1324) (quick hacky fix "Print settings -> G-code substitutions" F2900 for nothing, Take care your gcode only includes that feedrate in the wipe tower)

Also the speed multiplier set by the feature "Filament settings - >Wipe tower parameteres -> Max speed on the wipe tower" (that I believe was intended to be a fast work around to all this) resets the speed to 100% just after doing the infill instead of waiting for the perimeter and the support.

This speed multiplier creates the side effect:

> Filament max speed on wipe tower set to 400% violates maximum print speed rules (300mm/s)

since the feedrates are calculated normally with all the limits, THEN multiplied. The multiplier will break any limit by design.
Also interferes with manually adjusted speed multiplier during print, which is just unexpected and not pretty.

I agree that individual controls for speed on the wipe tower would be awesome, but also more difficult to implement.
From my view a fix for:
- Multiplier covering perimeters and supports
- Removing the hardcoded 2900 feedrate

would go a long way for the time being, until a way forward for handling all this and all its associated problems is found. 
(like custom gcode sections for the wipe tower, as ramming currently triggers filament sensors as they can't be disabled and re enabled, and others)
But that is more of a architectural decision needed from @supermerill on how to handle multimaterial and wipe towers rather than a bug or a feature.

neophyl, 2022-09-27:
One thing Prusa Slicer has always done is rely on the Max rates as set in a printers firmware (or sent to the printer via the M20x codes) to act as the ultimate upper limits on feedrates.  
Recently had a user on the Prusa forum whose test cubes were coming out with massively undersized Z and much thicker walls.  This was due to the Z feedrate using F6000 rather than the limit set in machine maximums.  His machine just couldn't do that feedrate on Z.   His old firmware/hardware board actually used the limits but his replacement bigteetech board/firmware didn't actually use any limits that were set.
So technically a firmware bug as it should respect them but it would be nice if PS checked those limits when generating gcode too so that they weren't in there in the first place.  I don't think it ever has.  

I's expect SuSi has many of the same niggles, although Merill has done a hell of a lot on trying to overcome some of them.