supermerill/SuperSlicer #2668: Paint-on supports don't respect camera perspective

Aquilux, 2022-04-08:
### What happened?

When using circle brush shape under paint on supports with "camera perspective mode" turned off (orthographic camera), painted "enforce supports", "block supports", and "remove selection" behave as though camera is in perspective mode.

Steps:
1. insert cube in new project
2. uncheck "use camera perspective" under preferences/camera
3. rotate camera so that target face is tilted slightly to be visible with edge aligned along view
4. use circle brush shape and click once on edge near barely visible face with brush extending across to paint on enforce/block or remove existing marker along entire face

[image]

Expected: parallel markings (distorted circle) spanning from one side of face to other, parallel to camera aligned edge

Actual: markings distorted in relation to center of view as if projected from camera with "camera perspective mode" turned on resulting in distinctly non-parallel makings and face detection issues that lead to incomplete/broken lines

[image]

[image]

Unrelated: project file instructions do not reflect current options. "Export project as..." does not exist under file > export, and no option to "export project" exists.

### Project file & How to reproduce

1. insert cube in new project
2. uncheck "use camera perspective" under preferences/camera
3. rotate camera so that target face is tilted slightly to be visible with edge aligned along view
4. use circle brush shape and click once on edge near barely visible face with brush extending across to paint on enforce/block or remove existing marker along entire face

### Version

Version 2.3.57 (latest release listed as stable as of submission)

### Operating system

Edition: Windows 10 ProVersion: 20H2 Installed on	‎7/‎22/‎2020 OS build	19042.1586 Experience	Windows Feature Experience Pack 120.2212.4170.0

neophyl, 2022-04-08:
It should be Save Project as, not export.

supermerill, 2022-04-09:
> rotate camera so that target face is tilted slightly to be visible with edge aligned along view

impossible to get without the perspective view. Or the line is very strait, as exepected.

Aquilux, 2022-04-14:
I should look into recording this behavior...

Done. [https://youtu.be/5iGBsJSimqc](Here is a link to video demonstrating the issue and steps to recreate.)

And here is the resulting project file.
[Shape-Box bug report.zip](https://github.com/supermerill/SuperSlicer/files/8485716/Shape-Box.bug.report.zip)

supermerill, 2022-04-14:
Can't do the same thing as you. The rotation is locked for me.
Are you using a 3D mouse?
[cant_paint_like_you.zip](https://github.com/supermerill/SuperSlicer/files/8487769/cant_paint_like_you.zip)

Aquilux, 2022-04-14:
Yes, but the basic mouse rotation should be enough. Let me demonstrate...
Man running a mouse visualizer + obs + superslicer really messed with things, the mouse visualizer generated a heavy lag so my inputs are a bit weird but here is the video: https://youtu.be/h49lC2ZzLGE

Aquilux, 2022-04-14:
> > rotate camera so that target face is tilted slightly to be visible with edge aligned along view
> 
> impossible to get without the perspective view. Or the line is very strait, as expected.
> 
> My test: (perspective in blue, the other in red) [image]

One thing to note, the effect becomes stronger the further from the center of the view port the brush is and how close the camera is to the part as the brush is still behaving as though "use camera perspective" is on even when it is off. The red line in this screen shot looks at first to be perfectly vertical, but it's actually at a slight angle. you can see this in the few polygons on it's right side at the top and how the right edge extends further than the left leaving the bottom end to look like it's rotated slightly to the left.