supermerill/SuperSlicer #2670: Ext. perimeter fanspeed overwrites overhang fanspeed

foxmckraut, 2022-04-09:
### What happened?

if you use -1 or 0-60% fanspeed at external perimeters, overhangs will be fanned fine. at 70% external perimeter fanspeed the overhangs will not blown at 100% anymore.

### Project file & How to reproduce

use external perimeter fan -1 > slice
use 80% > slice
[image]
[image]
[image]
[image]

### Version

2.4.58.2

### Operating system

Macosx 12.2.1

### Printer model

Anycubic i3 mega s

foxmckraut, 2022-04-11:
Oh sry my Laptop is in repair now I’ll share an Projekt file I have on my nas:

https://www.file-upload.net/download-14902192/7tailed_fox_supports1.3mf.html

This should do the job

supermerill, 2022-04-12:
it's the short layer time that kicks in and erase the override.
I'll look at how to fix that.

supermerill, 2022-04-12:
Ok,
first, you have to lower the overhang threshold or other is none. 30% is what you want, I think.

Then, the problem lie on the fan speedup time, that is also pushing the slowdown (it shouldn't). Using "only for overhangs" (or disable it) seems to resolve the issue.