supermerill/SuperSlicer #2671: Speed limits ignored for 2nd layer

jhlchu, 2022-04-09:
### Version
2.3.57.12

### Operating system type + version
Win10 Enterprise

### Behavior
Speed limit imposed by Filament Settings (e.g. Slow down or Target Layer Time) is ignored for 2nd layer only. Every other layer above it runs with speed limit. This is a problem for filaments that print hot (240+ C) or filaments that have shrinkage sensitivity due to inconsistent print speed.

* _Steps needed to reproduce the problem_
  * Slice any file with more than 5 layers.
  * Go to Gcode Preview with View>Speed
* _Expected Results_
  * Every layer except base layer should be limited by speed limit.
* _Actual Results_
  * 2nd layer (i.e. The first layer above the base layer) ignores speed limit while every other layer above it slows to speed limit, even if printing layer with the same area.
 
This file is just an example but this happens even with a simple cube. Basically any file I slice this is a problem. I can't find any settings to disable this.
  
[image]
**Base Layer**
[image]
**2nd Layer (the problem)**
[image]
**3rd Layer (should be running at this speed)**
[image]

[2nd Layer Speed Limit.zip](https://github.com/supermerill/SuperSlicer/files/8456434/2nd.Layer.Speed.Limit.zip)

jhlchu, 2022-04-12:
Not fixed for me in SS 2.4 or nightly.
I can trigger it consistently as well.

1. Use a system preset 'Printer Setting' and 'Print Setting'. I'm just using system preset 'Original Prusa i3 Mk3S and Mk3S+'. Print setting is system preset '0.2 mm Speed'
2. Change perimeter speed Internal and External to '150'. And turn off Skirt loop. [image]
4. This is result. 2nd layer speed difference from 3rd or even 4th. [image]

I can even get it to bug out on the 1st infill layer by changing 'Filament Settings'>'Layer Time Goal', from default 15 to 10.
[image]
[image]

It ignores the speed limit even though it is printing the same area (mm^2) as infill layers above it. Oddly, this causes the 2nd layer to obey speed limit again.

Here is the exact same thing from PS 2.4 from steps 1-4 from above (except for the change in Layer Goal Time).
[image]

[Shape-Box Speed Difference.zip](https://github.com/supermerill/SuperSlicer/files/8469785/Shape-Box.Speed.Difference.zip)

jhlchu, 2022-04-12:
I can confirm that as well. But it occurs at the 122/123 transition for me. Disabling 'Layer Goal Time' does make it all the same speed, but it's just trying to full speed every layer.

For me, I think I've narrowed it down to 'Infill Speed'>'Solid'. I retried the steps 1-4 I listed above but starting from system preset '0.20mm Quality' instead. The issue didn't occur, but once I switched it back to system preset '0.20mm Speed', and just transferred the Perimeter speeds, it occurred again. After changing each speed and testing, dropping Solid Infill speed < 100 made is disappear.

[image]
[image]

supermerill, 2022-04-13:
can't reproduce any issue with 2.4.58.2:
[image]

jhlchu, 2022-04-14:
It's visible on the image you posted. Bed layer ~16.7mm/s, 2nd layer ~150mm/s, 3rd and 4th layer ~100mm/s. 5th ~33.4 (infill start?), so I'm guessing 4 bottom layers? 2nd, 3rd, 4th should be same speed if they are all solid fill with same area. No reason 2nd layer should be suddenly speed up +50mm/s.

supermerill, 2022-04-14:
The layer time of the second layer is the same as the third and fourth, so that's why.
The question may be why the computed layer time is the same?

jhlchu, 2022-04-14:
How many bottom solid fills were in the your example? I was assuming 4 bottom (including bed layer). If they took the same time to cover the same amount of area (mm^2), I'd thought they'd run at the same speed (except first) since they use same perimeter and solid fill settings. Or like you said, if they running at +50mm/s difference between 2nd and 3rd+4th, why is the computed layer time the same?

In your example, is the 2nd layer infill running slower than the 3rd layer infill, so computed layer time comes out the same? Although it becomes harder to tell what the actual value is when the colors are so close.

supermerill, 2022-04-14:
> How many bottom solid fills were in the your example? 

it's your project

btw, after many hours lost, here is what I can explain:
 - the speed is unobtainium, it can't reach it with the acceleration set. So to increase the time per layer, is has to decrease it a lot.
 - there is a crude mechanism to avoid too sudden change from layer to layer, and it works by averaging with half of the previous layer time.

jhlchu, 2022-04-15:
If it's the my example from above, then it was indeed 4 layer bottom.
Is this just an edge case of mismatched settings or a bug?
Is there a way to toggle the half previous layer time averaging setting to turn it off?

Thanks for taking the time taken to look into this, I appreciate it.

supermerill, 2022-04-15:
> Is this just an edge case of mismatched settings or a bug?

not a bug because i'ts behaving as expected, even if the behavior isn't the one you want. It's more of an improvement request.

> Is there a way to toggle the half previous layer time averaging setting to turn it off?

no kill switch yet.

jhlchu, 2022-04-17:
> no kill switch yet.

I'd like to make of feature request of this then, please.

engtauan, 2022-07-10:
I solved deactivating overhang detection. My speed for bridge was superior than my perimeter speed. When the first layer compensation act the overhang angle become too steep and the overhang detection turn the speed up.