supermerill/SuperSlicer #2673: Paint-on seam's paint move triggers slicing.

Lecso11, 2022-04-09:
### What happened?

After painting on seam lines, the slicer starts slicing right after I finish a paint move (stop holding left mouse button).
It is very annoying, as I finish a paint move, the window jumps to sliced view and closes seam painting tool.

### Project file & How to reproduce

[a_drive_frame_lower.zip](https://github.com/supermerill/SuperSlicer/files/8456585/a_drive_frame_lower.zip)

### Version

2.4.58.2

### Operating system

Windows 10

### Printer model

Voron 2.4 with klipper

neophyl, 2022-04-09:
Turn off background processing in Configuration>Preferences>General>Automation.  This is a program level setting and not stored in a project.  As such your project doesnt automatically slice when painting seams for me as I don't ever have background processing enabled.

Lecso11, 2022-04-09:
> Turn off background processing in Configuration>Preferences>General>Automation. This is a program level setting and not stored in a project. As such your project doesnt automatically slice when painting seams for me as I don't ever have background processing enabled.

It is off for me too.

supermerill, 2022-04-11:
seems to only happen if the object with the current settings has already been sliced.

closed 2024-02-11 by supermerill