supermerill/SuperSlicer #2675: Mac M1

drveni, 2022-04-09:
### What happened?

wont start

### Project file & How to reproduce

SuperSlicer_2.3.57.12_macos_220401 open correct and start

SuperSlicer_2.4.58.2_macos_220401 wont start...

### Version

2.4.58.2

### Operating system

M1 mac

### Printer model

non

foxmckraut, 2022-04-09:
try https://github.com/supermerill/SuperSlicer/issues/2543

supermerill, 2022-04-10:
can you test this build for macos arm? https://mega.nz/file/9gwyHYKS#-BJPvMGzupZPLbnMqWBF3elMV7BLzXXqnBZ6IXNO2HU
I tried with zstd uninstalled, so it shouldn't be needed, but I'm not 100% confident.
i guess it still need the xattr thingy to be launch on another computer than the one that built it?

drveni, 2022-04-11:
> can you test this build for macos arm? https://mega.nz/file/9gwyHYKS#-BJPvMGzupZPLbnMqWBF3elMV7BLzXXqnBZ6IXNO2HU I tried with zstd uninstalled, so it shouldn't be needed, but I'm not 100% confident. i guess it still need the xattr thingy to be launch on another computer than the one that built it?

it works. I started. I'll try to test later slice and menu. i don't have zstd installed. xattr thingy im not shure what it is I just ran over the old last one that worked.

Juggeli, 2022-05-02:
> can you test this build for macos arm? https://mega.nz/file/9gwyHYKS#-BJPvMGzupZPLbnMqWBF3elMV7BLzXXqnBZ6IXNO2HU
> I tried with zstd uninstalled, so it shouldn't be needed, but I'm not 100% confident.
> i guess it still need the xattr thingy to be launch on another computer than the one that built it?

This has been running great. I have sliced about dozen models for several printers without any hiccups. It needed the xattr command to run tho.

supermerill, 2022-05-02:
latest nightly:
https://mega.nz/file/NxIigRwR#hKgPdD1LykfLO3cwrUt_Qzs4BvcDVgh94_KDSNcV788

drveni, 2022-05-02:
> latest nightly: https://mega.nz/file/NxIigRwR#hKgPdD1LykfLO3cwrUt_Qzs4BvcDVgh94_KDSNcV788

i gotta problem with this after dowlonad and install. startup app says Superslices is damaged and cant be opened... you should move it to the bin

foxmckraut, 2022-05-02:
> > latest nightly: https://mega.nz/file/NxIigRwR#hKgPdD1LykfLO3cwrUt_Qzs4BvcDVgh94_KDSNcV788
> 
> i gotta problem with this after dowlonad and install. startup app says Superslices is damaged and cant be opened... you should move it to the bin

do the xatr thing...

try:
sudo xattr -cr "/Volumes/MacOS SSD/Applications/SuperSlicer.app"
or

sudo xattr -cr "/Volumes/Macintosh HD/Applications/SuperSlicer.app"

in terminal

drveni, 2022-05-02:
> sudo xattr -cr "

sorry nothing is happened....

first download liked gread..  dragged the file where i want to install clicked and launched..

foxmckraut, 2022-05-02:
> > sudo xattr -cr "
> 
> sorry nothing is happened....
> 
> first download liked great.. dragged the file where i want to install clicked and launched..

if you drag it anywhere else than to aplication you need to correct the path i posted above.

drveni, 2022-05-02:
> > > sudo xattr -cr "
> > 
> > 
> > sorry nothing is happened....
> > first download liked great.. dragged the file where i want to install clicked and launched..
> 
> if you drag it anywhere else than to aplication you need to correct the path i posted above.

yes i did that....yes I know that and I understand ... looks some logs from mac .. and I don't see anything special.

foxmckraut, 2022-05-02:
download > drag to aplication > sudo xattr -cr "/Volumes/Macintosh HD/Applications/SuperSlicer.app" > launch SS

drveni, 2022-05-02:
> download > drag to aplication > sudo xattr -cr "/Volumes/Macintosh HD/Applications/SuperSlicer.app" > launch SS
> 
> correct?

yes!!

from terminal 

tihomir@Tihomirs-Mac-mini SuperSlicer.app % pwd
/Users/tihomir/Desktop/SuperSlicer.app
tihomir@Tihomirs-Mac-mini SuperSlicer.app % sudo xattr -cr "/Users/tihomir/Desktop/SuperSlicer.app"          
Password:
tihomir@Tihomirs-Mac-mini SuperSlicer.app %

foxmckraut, 2022-05-02:
seems correct but i cant say if it will work on desktop...you tried also in aplications folder?

thats i only thing i could say is different between your mac an mine...

drveni, 2022-05-02:
> seems correct but i cant say if it will work on desktop...you tried also in aplications folder?
> 
> thats i only thing i could say is different between your mac an mine...
[image]

installed to aplication folder... and same error...

foxmckraut, 2022-05-02:
installed in aplication followed by sudo xattr -cr "/Volumes/Macintosh HD/Applications/SuperSlicer.app"  in terminal correct?

you can also try pressing option button+leftclick and than open.

drveni, 2022-05-02:
> 

my option button + right click nothinh opened... when i enter in superslicer.app i see all directoy and subdirectoys... and files.. all seems ok 

are you corrected upload files? try download 3 times same file.. same error.. im not shure.. but first lik what workend instaled od another mount drive and works... this gotta error. i really dont know. what log to look?? is there another log file somewhere? I don't see anything from the mac?.

foxmckraut, 2022-05-02:
> my option button + right click nothinh opened... when i enter in superslicer.app i see all directoy and subdirectoys... and files.. all seems ok
>
> are you corrected upload files? try download 3 times [...]

i have an macbook air 2020 m1 and downloaded it from the link above...done the things i told you and it worked properly...

drveni, 2022-05-02:
> > > 
> > 
> > 
> > my option button + right click nothinh opened... when i enter in superslicer.app i see all directoy and subdirectoys... and files.. all seems ok
> > are you corrected upload files? try download 3 times same file.. same error.. im not shure.. but first lik what workend instaled od another mount drive and works... this gotta error. i really dont know. what log to look?? is there another log file somewhere? I don't see anything from the mac?.
> 
> i have an macbook air 2020 m1 Macosx 12.3.1 and downloaded it from the link above...done the things i told you and it worked properly.. [image] .

im on Mac mini (M1, 2020) 12.3.1.. i realy dont know why