supermerill/SuperSlicer #2676: Android build of SS?

staffert, 2022-04-09:
**Is your feature request related to a problem? Please describe.**
Is there a build of SuperSlicer for Android devices?

**Describe the solution you'd like**
I would like to be able to run SS on my Samsung Galaxy TAB S7+ tablet (or similar)

**Describe how it would work**
Just thought it would be convenient and nice to be able to work on slicing/settings while wireless.
With network storage and cloud support, people could esily prepare print files on the go anywhere, then launch prints to their network connected printers.

**Describe alternatives you've considered**
Laptops are clunky and heavy at best, not always convenient.
Optional Touch UI tweaks would make it a rocking app with very little changes.

supermerill, 2022-04-11:
Problem is, I don't know by how much the GUI is useable on a touch-only device.
And changing it to be usable, or even adapted for it, is very very costly (in time)

staffert, 2022-04-11:
What if you just let the Android rap with the deets and just provide a build?
Quite frankly my tablet support full keyboard and mouse input. I'm fine with that.