supermerill/SuperSlicer #2678: Flip build table axis

SiCFPV, 2022-04-10:
Well, I've searched a lot about it, but couldn't find nothing. 

My 0 is at the right side of the bed, and there's no way to flip the sides, it stays only at the left side and my prints got mirrored, is it possible to fix this? This is a problem for me and due to this I have to change slicer, and that's not what I want after being working on SuperSlicer for 5 months.

Thank you very much. Apprecieted

supermerill, 2022-04-10:
In your bed shape.
Set the same value in `origin` as in `size`

SiCFPV, 2022-04-11:
> In your bed shape.
> Set the same value in `origin` as in `size`

I've tried, got always moving errors on Klipper and wants to print always on the right side not matter origin I add.

I'm doing mirror on all my pieces.

neophyl, 2022-04-11:
Isn't this something you can configure in your Klipper setup ?  I know normally its something that can be set when you compile the firmware for the printer under say Marlin for example.  

I'm surprised that a new and flexible setup like Klipper cant do this.

SiCFPV, 2022-04-11:
> Isn't this something you can configure in your Klipper setup ?  I know normally its something that can be set when you compile the firmware for the printer under say Marlin for example.  
> 
> I'm surprised that a new and flexible setup like Klipper cant do this.

It's not Klipper, but SuperSlicer like I've wrote! You can see here with Simplify3D, and now try what advised above on SuperSlicer

[IMG_20220410_112550.jpg](https://user-images.githubusercontent.com/76414558/162688257-232e7d60-5d48-4e52-a43c-cabc49071e6d.jpg)

supermerill, 2022-04-11:
> I've tried, got always moving errors on Klipper and wants to print always on the right side not matter origin I add.

ok, I understand the problem.
What you want is currently not supported.

SiCFPV, 2022-04-11:
> > I've tried, got always moving errors on Klipper and wants to print always on the right side not matter origin I add.
> 
> ok, I understand the problem.
> What you want is currently not supported.
> 

Thanks in Advance! For the moment I fixed with mirroring the X axis, works great, but not user friendly 😂

[image]