supermerill/SuperSlicer #2680: OSX SS 2.4.x Gcodeviewer Crash on startup

OstroMa, 2022-04-10:
OSX Gcodeview crash on startup

Branch: Nightly_dev

OS: OSX Intel and ARM64

https://github.com/supermerill/SuperSlicer/blob/8df1a06056bdd583c80d0cf77028768f4fe6b00a/src/slic3r/GUI/MainFrame.cpp#L152

Icon SuperSlicer-gcodeviewer-mac_128px.png is missing.

supermerill, 2022-04-10:
can you test this build for macos arm? https://mega.nz/file/9gwyHYKS#-BJPvMGzupZPLbnMqWBF3elMV7BLzXXqnBZ6IXNO2HU
did you have zstd installed (shouldn't be needed from my test)?

OstroMa, 2022-04-11:
i have libzstd installed.

yes this version is working.

i must execute:
sudo xattr -cr "/Volumes/MacOS SSD/Applications/SuperSlicer.app"
before.

normally i compile my on version.

closed 2022-05-07 by OstroMa