supermerill/SuperSlicer #2682: Settings search does not work properly

bassamanator, 2022-04-11:
### What happened?

The search is mostly broken. I thought I was doing something wrong so I tested it in PrusaSlicer, but the search works there.

[image]

### Project file & How to reproduce

1. Go to any settings page.
2. Search for 'spiral'
3. Spiral vase setting will not be found

Expected behaviour:
-Spiral vase setting should show up at the top of the search results.

[image]

### Version

Version 2.3.57 SuperSlicer-ubuntu_18.04-gtk2-2.3.57.12.AppImage

### Operating system

5.15.28-1-MANJARO

### Printer model

Voron 0.1 with mini-sherpa; default profile mixed with Ellis profile.

supermerill, 2022-04-11:
can't reproduce on windows.
maybe duplicate of #2672

bassamanator, 2022-04-11:
>if you click on the first entry, where do you go?

Oh it actually takes me to the *expected* results, that is, searching for 'spiral', shows incorrect results, however, the first entry leads me to 'Spiral vase' entry.

supermerill, 2022-04-11:
so it's really a duplicate of 2672, as it seems to have a refresh issue on gtk2

bassamanator, 2022-04-11:
Don't know much about gui toolkits but I just tried [SuperSlicer-ubuntu_18.04-2.3.57.12.AppImage](https://github.com/supermerill/SuperSlicer/releases/download/2.3.57.12/SuperSlicer-ubuntu_18.04-2.3.57.12.AppImage) and it was the same problem there.

closed 2024-08-27 by bassamanator