supermerill/SuperSlicer #2683: Need X/Y IDEX Calibration object

inspectionsbybob, 2022-04-11:
### Version
2.3.57

### Operating system type + version
Win10

### Behavior
Need a tool/Calibration object for aligning IDEX extruders

_Is this a new feature request?_
YES

This  is a request for a X/Y IDEX calibration object set similar to CURA's 

I tried to export the Object from CURA as a 3MF but it comes across at a weird height and won't let me adjust it onto the Platform, it looks warped as it won't print properly and when I look at the underside it looks like it has a step in it.  In any case I need a SS version of this object

[image]

#### Project File (.3MF) where problem occurs
[Calibration X-Y.zip](https://github.com/supermerill/SuperSlicer/files/8464777/Calibration.X-Y.zip)

The Object should have TWO parts that look like Combs, and where they line up tells you how much to adjust your offsets, either in software or in the Printer (preferred).  They should be only a few layers thick so they print FAST.

Prusa has something different on their site at:
https://www.printables.com/model/138740-dual-extruder-calibration-model
or
https://www.printables.com/model/129617-offset-xy-dual-extruder-idex-calibration

I have tried a number of these and they print weird, and even when I send the to the Gcode preview app they look weird...

HELP Please as I need this to get my printer aligned...

supermerill, 2022-04-11:
can't see anything wrong with your first one. there are comb, and i can move them by a specific offset relative to each other. They also are thin.

[image]

I don't understand your problem.

inspectionsbybob, 2022-04-12:
I found some of the issues... a z-offset that had been set from the instal.
I didnt set it as this was a clean install and I haven't used the IDEX
function on the printer before.

I still would like to see this, or something like it added to your
calibration routines...

Has some issues with the purge tower, but I think most of it is settings.

It didnt like my position, said something about having to use relative
positioning to use the purge??

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2022-04-12:
Sorry, related question... got it CLOSE with original file I sent, went to
use a "better" file to get the last .1 mm dialed in.
It won't let me change the Z-position of the overlay.
I tried dragging it up, it snaps back to .2
I tried editing the value in the Box, it snaps back to .2

I guess I don't understand the size and position, I am used to Cura...

The Base should be flat on the platform, which I am used to being "0"  and
then the height would be the thickness, lets assume it is .4 to the top and
another .4 for the bars

The OVERLAY should then be at a height of .4 so that it Prints ON TOP of
the base....

But BOTH are starting at a non-zero height, and won't let me adjust them
[image: Screenshot (24).jpg]

I am missing something basic here....

The Base layer is
Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
***@***.***
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

inspectionsbybob, 2022-04-12:
Here are the files, when I responded it didn't send the attachments.
[Offset_X_Y_dual_extruder_2.zip](https://github.com/supermerill/SuperSlicer/files/8474631/Offset_X_Y_dual_extruder_2.zip)
[image]

PS... when I brought this into CURA it behaved as expected, I could change the heights and then Merge( align) them and print.

supermerill, 2022-04-13:
you have to add the overlay as a part of the other object.
Or if you imported both at the same time, the slicer can ask you if it has to combine them.
You can also select both in the right panel and use "merge" from he contextual menu.

inspectionsbybob, 2022-04-13:
Will try that after my big print finished in 8hrs or so.

Still would like to see some IDEX tools added to the calibration routines, please.

Would like to see a flat (x/y) calibration object and a
3D object, typically a 3D cube added.

supermerill, 2022-04-13:
I have the support calibration, the pa calibration, if I have some time to spare.

I'll see if I can add a way for users to create their calibrations.

cryoglobe, 2023-01-16:
Hello @inspectionsbybob    
this calibration test is great .... simply import both files in the same time and combine them :)   I use it all the time with my IDEX Sv04
https://www.printables.com/fr/model/129617-offset-xy-dual-extruder-idex-calibration