supermerill/SuperSlicer #2685: SuperSlicer crashes the whole system

tapir, 2022-04-11:
### What happened?

In certain conditions, slicing action can crash the whole system.
I'm on arch linux with latest radeonsi drivers. I have tried both wayland and xorg and it didn't make any difference.
I create a test cube from the menu. It automatically slices it. Then I change some settings and re-slice manually. After trying re-sliceing 1 or 2 times, without any exception, slicing progress bar freezes. The whole screen freezes. I get a black screen and when it comes back there are dots/pixels of mixed color all over the screen. I can still move the mouse cursor but can't click anything thus I'm not able to get check any logs. I can't use the system without a hard reset.

### Project file & How to reproduce

Standard test cube or voron cube can be used to reproduce it.

### Version

Both 2.3 and 2.4 latest versions

### Operating system

Arch Linux

### Printer model

Voron 2.4 250mm

supermerill, 2022-04-11:
I'll try to "slice a cube and change parameters" on my ubuntu VM. If I can't reproduce, I wcan't fix that.
And I don't have time to support all linux distro.

Try the gtk2 if you're on gtk3, or the opposite.

tapir, 2022-04-11:
I understand. Is there any way for me to get some logs? Where does SuperSlicer outputs stuff? I tried running from the terminal but it's empty.

In the meanwhile, here how it looks like https://imgur.com/a/RFVGfws
It's not just the superslicer window, all windows are same and unresponsive.

I tried both GTK2 and GTK3, appimage and ubuntu packages. 2.3 and 2.4. Same always. PrusaSlicer on the other hand works OK.

Also tried the Windows version, and that seems to work fine too.

supermerill, 2022-04-11:
> Is there any way for me to get some logs? Where does SuperSlicer outputs stuff? I tried running from the terminal but it's empty.

You can have logs by launching it with `-loglevel=5` (5 is the most verbose, if you only want errors, I think 1 is enough)

tapir, 2022-04-11:
For what it's worth I've found this on kernel log
```
Apr 11 20:04:02 anarchy kernel: [drm:amdgpu_dm_atomic_commit_tail [amdgpu]] *ERROR* Waiting for fences timed out!
Apr 11 20:04:02 anarchy kernel: [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx_0.0.0 timeout, signaled seq=3493, emitted seq=3496
Apr 11 20:04:02 anarchy kernel: [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process slic3r_main pid 1549 thread superslice:cs0 pid 1598
Apr 11 20:04:02 anarchy kernel: amdgpu 0000:03:00.0: amdgpu: GPU reset begin!
Apr 11 20:04:02 anarchy kernel: amdgpu 0000:03:00.0: [drm:amdgpu_ring_test_helper [amdgpu]] *ERROR* ring kiq_2.1.0 test failed (-110)
Apr 11 20:04:02 anarchy kernel: [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* KGQ disable failed
Apr 11 20:04:02 anarchy kernel: amdgpu 0000:03:00.0: [drm:amdgpu_ring_test_helper [amdgpu]] *ERROR* ring kiq_2.1.0 test failed (-110)
Apr 11 20:04:02 anarchy kernel: [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* KCQ disable failed
Apr 11 20:04:02 anarchy kernel: [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* failed to halt cp gfx
```
looks like slic3r_main crashes the gpu driver

K1llf0rce, 2022-05-02:
Any news on this? I have been having the same issue and I couldn't figure out the cause yet.

supermerill, 2022-05-02:
problem with the installed glew/opengl dirvers, can't do anything, as I don't have the exact config.

K1llf0rce, 2022-05-02:
Yeah, so I found some time again to look into this today and it really seems like it is unrelated to SuperSlicer.

I get the same error as @tapir above:
```
May 02 20:15:18 pcft kernel: [drm:amdgpu_dm_atomic_commit_tail [amdgpu]] *ERROR* Waiting for fences timed out!
May 02 20:15:24 pcft kernel: [drm:amdgpu_dm_atomic_commit_tail [amdgpu]] *ERROR* Waiting for fences timed out!
May 02 20:15:24 pcft kernel: [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx_0.0.0 timeout, signaled seq=6803, >
May 02 20:15:24 pcft kernel: [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process slic3r_main pi>
May 02 20:15:24 pcft kernel: amdgpu 0000:28:00.0: amdgpu: GPU reset begin!
May 02 20:15:24 pcft kernel: amdgpu 0000:28:00.0: [drm:amdgpu_ring_test_helper [amdgpu]] *ERROR* ring kiq_2.1.0 tes>
May 02 20:15:24 pcft kernel: [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* KGQ disable failed
May 02 20:15:25 pcft kernel: amdgpu 0000:28:00.0: [drm:amdgpu_ring_test_helper [amdgpu]] *ERROR* ring kiq_2.1.0 tes>
May 02 20:15:25 pcft kernel: [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* KCQ disable failed
May 02 20:15:25 pcft kernel: [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* failed to halt cp gfx
```

And after googling and browsing through many threads it seems this is an issue in the amdgpu driver that has apparently been present for a long time. Weirdly enough it doesn't seem to affect Laptop iGPU's, which I also confirmed because SuperSlicer works perfectly smooth on my VEGA graphics based Laptop.

Seems to be this: https://gitlab.freedesktop.org/drm/amd/-/issues/934
And also leaving this here: https://forum.manjaro.org/t/system-frequently-crashing-after-gpu-drivers-update/62139/205

K1llf0rce, 2022-05-07:
I have now isolated the issue. It seems to be an issue in the `radeonsi` driver, which is the OpenGL driver for AMD cards. It's included in the `mesa` package, so until a fix arrives, here is a workaround.
Use a different OpenGL driver, for example `zink` which uses Vulkan API calls and is GPU architecture independent.

Start superslicer with the following environment variable:
```
MESA_LOADER_DRIVER_OVERRIDE=zink superslicer
```
It should then open and slice models without crashing.
I see this issue as closed since it isn't directly related to SuperSlicer.

closed 2022-05-08 by tapir