supermerill/SuperSlicer #2687: Seam Position Error

Garryb123, 2022-04-11:
A path starts on a corner when I had set the seam to 'rear'.

I expected all the loop starts to be in the rear and all were except this one.

[RFS_Sw_Jack_Mount.zip](https://github.com/supermerill/SuperSlicer/files/8467701/RFS_Sw_Jack_Mount.zip)

Windows 10
SS Version 2.3.57
Home made printer running Duet FW

[image]

supermerill, 2022-04-12:
seems fixed in 2.4
Won't fix the 2.3.

Jamin82, 2022-04-13:
In the latest nightly I am getting odd seam behaviour:

if i set the seam position to rear i get the expected behaviour:
[image]

if i manually set seam position it completely ignores the position of the orb
[image]

[image]

[seam position error.zip](https://github.com/supermerill/SuperSlicer/files/8481337/seam.position.error.zip)

supermerill, 2022-04-13:
> In the latest nightly I am getting odd seam behaviour:

found the bug (from the merge)
will be fixed in next nightly

closed 2022-07-14 by supermerill