supermerill/SuperSlicer #2689: Deretraction inside the print

jaimebpg, 2022-04-12:
When Printing with "external perimeter first" to improve surface finish and/or dimensional accuracy. Deratractions become an issue, they generate blobs when printing multiple parts and traveling thru open spaces.

I propose something like this:
lower nozzle inside the print -> deretract -> wipe -> lift nozzle and start extruding perimeter

supermerill, 2022-04-13:
note to myself: like the little wipe that go inside at the end of the extrusion, but at the start.

royal32, 2022-04-13:
Seconding this request! I've been running my own fork that does exactly this for a while now but my implementation is.... hacky at best. A properly coded implementation would be awesome.

https://github.com/royal32/SuperSlicer/commit/ecf49d575adf31f52b28de0133de98cf6a54228f

supermerill, 2022-04-13:
@royal32 If I have to do it, i'll do mostly the same as you. Just use the coPercent type for percentage-only settings.

Is the "depth" factor really that useful?

royal32, 2022-04-13:
The depth factor was for trying to tune for the best results without editing the code and recompiling every time, I just keep it at 5. too shallow and it doesn't have much of an effect, too "deep" and it can deretract outside of the part on thinner geometry. The exposed setting is nice to have but probably unnecessary. Yeah IIRC it's not exactly a percentage in the context of the actual equation. The percent sign is just because I copied the "Better bonding" setting.