supermerill/SuperSlicer #2690: Supports weird behavior control.

levleon13, 2022-04-13:
### Version
Latest

### Operating system type + version
WIN11 

### Behavior
Supports are being created in a little wider area than the area needed to be supported, creating tiny bits that 100$ fail on print.
Happens in any case, i am always fighting these, is there a way to tell the support that it should be ONLY directly under the exact area of the bridge\overhang?

[image]

supermerill, 2022-04-13:
The 2.4 versions have new settings to control that.

levleon13, 2022-04-13:
> The 2.4 versions have new settings to control that.

uhmm.. so this should be on the latest superslicer(2.3.57) right? what is the name of the setting?

supermerill, 2022-04-13:
> should be on the latest superslicer(2.3.57) right?

nop, it's on the beta ones. (2.4.58)

levleon13, 2022-04-13:
> > should be on the latest superslicer(2.3.57) right?
> 
> nop, it's on the beta ones. (2.4.58)

So i guess it will be implemented soon?

supermerill, 2022-04-13:
It's already implemented in the beta releases, available in the release page.

levleon13, 2022-04-13:
> It's already implemented in the beta releases, available in the release page.

Thank you!

closed 2022-07-14 by supermerill