supermerill/SuperSlicer #2696: [BUG] Fan speed problem

javales74, 2022-04-13:
### What happened?

When I set a fan speed to "support material interface", the slicer insert a default speed with no extrusion role change and overwrite the selected speed.

Sorry my english!

### Project file & How to reproduce

[image]
[image]

### Version

2.4.58.3

### Operating system

Windows 10

### Printer model

Ender 3 Pro

supermerill, 2022-04-13:
> Project file

? where?

javales74, 2022-04-13:
> > Project file
> 
> ? where?

javales74, 2022-04-13:
> > Project file
> 
> ? where?

Using Custom G-Codes or using G-Code Substitution, it's the same

supermerill, 2022-04-13:
I mean, you didn't include your porject file, so I can't test.

javales74, 2022-04-13:
[MaradonaBusto.zip](https://github.com/supermerill/SuperSlicer/files/8484652/MaradonaBusto.zip)
It´s not exactly the same project but you can view the same behaviour at Z 46,80 between the 251976 and 251994 lines.

Thanks!

supermerill, 2022-04-29:
can't reproduce on my latest nightly

[image]