supermerill/SuperSlicer #2697: No object option for "enforce_full_fill_volume"

TommyC81, 2022-04-13:
### What happened?

Select an object, "Add settings -> Perimeters Shell", there is no option to adjust "enforce_full_fill_volume".

### Project file & How to reproduce

Select an object, "Add settings -> Perimeters Shell", fail to locate an option to adjust "enforce_full_fill_volume".

Available first layer options:
[image]

### Version

nightly_dev (e941089c075efbdbf7f36959f868ce2b5ee4c633)

### Operating system

Windows 10

### Printer model

Creality CR6 with Sherpa Mini direct drive

supermerill, 2022-04-13:
it's currently under infill category
I'll move it.

TommyC81, 2022-04-13:
Thanks, I just located it in the "Add settings" menu option. I think it may actually be more logic to keep it under the "Infill" tab, that's what it is after all (rather than "Perimeters & Shell" - it's not really a perimeter nor shell item...).

Maybe somewhere here would make sense (or some other place in the "Infill" tab):
[image]