supermerill/SuperSlicer #2698: Various object "Add settings" typos

TommyC81, 2022-04-13:
### What happened?

Various typos in object "Add settings".

Also noted that the "settings" windows themselves are rather small (by default), it would be of benefit to have a larger window by default, and perhaps also include the exact variable name behind the description (for clarity).

### Project file & How to reproduce

Select "Add settings" for an object, locate the following typos/inconsistencies (words used, capital letters or not):
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Note: This is not necessarily a complete list, just some items that stood out as I was doing some print testing.

### Version

nightly_dev (e941089c075efbdbf7f36959f868ce2b5ee4c633)

### Operating system

Windows 10

### Printer model

Creality CR6 with Sherpa Mini direct drive

supermerill, 2022-04-13:
It's currently using a built-in dialog. To change its appearance (for adding tooltips for example), I need to rewrite it.

TommyC81, 2022-04-14:
> It's currently using a built-in dialog. To change its appearance (for adding tooltips for example), I need to rewrite it.

Thank you for considering, it is not worth the effort to undertake at this time. Please disregard.

Meanwhile, found another "Add settings" inconsistency (should be "Solid fill pattern", or possibly "Solid infill pattern" - the parameter is "solid_fill_pattern"):
[image]

supermerill, 2022-04-14:
fixed with `Solid fill pattern`
also changed `Top pattern` to `Top fill pattern`
and `Fill pattern` to `Sparse fill pattern`
for consistency

supermerill, 2022-04-14:
What are your proposal for the other ones?
Min gap-fill surface -> Min surface for gap filling?
?

supermerill, 2022-04-14:
If you are able to create a pull request, then great, I'm waiting it.
Note that the categories are only used for the GUI menu, so you can add more if you want (don't forget the icon in GUI_Factories.cpp:l123 and the label in Config.cpp)