supermerill/SuperSlicer #2700: s_wall_thickness Buggy in 2.4.58.2

adawalli, 2022-04-13:
**Bug description**
The s_wall_thickness parameter cannot be adjusted as expected. When set to any value and then saved - closing/reopening SS...the value changes to a new random value.

**Expected behavior**
I expect this to stay fixed

[wall_thickness.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8484176/wall_thickness.3mf.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Print Settings
2. Adjust wall thickness to 0
3. Save Profile
4. Close SS
5. Reopen SS and select profile - witness new weird wall thickness that doesn't match what you saved

This was first noticed in https://github.com/AndrewEllis93/Ellis-PIF-Profile/issues/7

**Desktop (please complete the following information):**
 - OS: Mac Monterey
 - Version : 2.4.58.2 (x86 variant)
 - Printer model: Voron 2.4

supermerill, 2022-05-05:
now (next nightly),
* if the value is invalid (too low), it will revert to saved values.
* if 0, it will revert to saved values but the number stay at 0 (kind of a fake 'disabled', as it does nothing by itself, so nothing is disabled)
* 
it should be enough?

adawalli, 2022-05-05:
works for me - thanks!

closed 2024-02-11 by supermerill