supermerill/SuperSlicer #2701: Klipper calibration prinnts requested

Eifel-Joe, 2022-04-13:
**Is your feature request related to a problem? Please describe.**
It would be nice to have the standrad Klipper calibration prints in the Calibration Menu.

**Describe the solution you'd like**
Add the Klipper square_tower.stl for calbration Pressure Advance and the Klipper ringing_tower.stl for calibration of Input Shaper. 

**Describe how it would work**
It should just work like the Bed Leveling calibration. Just choose the new calibration and the stl is loaded on the print surface.

**Describe alternatives you've considered**
Of course one could save this test as 3mf on harddisk, but you could also do this with all of these calibration prints. ;-) 

**Additional context**
I hav no additional context.

supermerill, 2022-04-13:
duplicate of #320 #282

foxmckraut, 2022-04-16:
Ringing tower: https://www.klipper3d.org/Resonance_Compensation.html
STL: https://www.klipper3d.org/prints/ringing_tower.stl

Presure advance: https://www.klipper3d.org/Pressure_Advance.html
STL: https://www.klipper3d.org/prints/square_tower.stl
Gcode Command: SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=1 ACCEL=500
for bowden:  TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.020
for DD: TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.005

Both Vasemode 

The Presure advance tower should also work on Marlin and Reprap but i didnt know the gcode commands for them.

i think the pa tower is more interesting cause your input shaper didnt change very often but pa can change on every filament.