supermerill/SuperSlicer #2702: Bug/Features  filament_colour only exports as main colour, not counting any color changes

DatenThielt, 2022-04-13:
**Bug description**
With a print that has had several color change steps added,  when exporting the Gcode M600 {filament_colour}  all colors are reported as the filament color selected in filament settings, and does not take into account any M600 steps added.  whilst this can be done using the custom gcode, it does not then conform with the interface showing an accurate representation 

**Expected behavior**
on export {filament_colour} should represent the current color that has been switched to or will be switched to rather than the filament itself, This would then allow interfaces to display what filament color it is expecting to be loaded

**To Reproduce**
Steps to reproduce the behavior:
1. set Filament Change gCode to M600 {filament_colour}
2. Import object and set multiple colors using the interface
3. Export the gCode and view it, 
4. Observe that all M600 commands are sent with the base colour of the filament and do not change as expected

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - 
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/8485245/Shape-Box.zip)

 - Version [e.g. 2.4.58.2]
 - Voron V 2.4 Running Klipper

supermerill, 2022-04-14:
As the tooltip says, if you're using your own toolcahnge gcode, add a T{next_extruder} (or ACTIVATE_EXTRUDER EXTRUDER={next_extruder}) if you want the gcode viewer to understand that the extruder has changed.

DatenThielt, 2022-04-14:
is next_extruder a special variable or am I being dumb?  It does not appear to be valid,  Also I think I would add ACTIVATE_EXTRUDER EXTRUDER={next_extruder} before the M600 command correct?

supermerill, 2022-04-14:
oh sorry, i misunderstood the issue.
`filament_colour` is the colour defined in `Filament Settings ->Color`
colours from custom "color print" aren't available

DatenThielt, 2022-04-19:
That's great! Hopefully its not too much of a pain but I can certainly see it being useful.  And I noticed color and color everywhere, I cant decide myself which ones to use! (More partial to Color but eh, what can you do)

closed 2024-02-11 by supermerill