supermerill/SuperSlicer #2707: Split is disabled

StephenRC, 2022-04-14:
### Version
2.4.58.2

### Operating system type + version
Win 10 x64

### Behavior
Split parts not available on right click popup menu
Load any multi part stl, right click on stl or stl name, split is disabled.

supermerill, 2022-04-14:
a stl is only one object.
If your human eye can cut it, please do so manually as the software don't have the algorithm to guess it right now.

StephenRC, 2022-04-14:
PrusaSlicer 2.4.0+win64 and 2.4.1+win464 can split the stl files.

supermerill, 2022-04-14:
Can you give me an exemple?

jhlchu, 2022-04-15:
@StephenRC Do you mean the 'Context Menu'>Split?
SS 2.3.57
[image]
SS 2.4
[image]
PS 2.4.2-rc2
[image]

supermerill, 2022-04-15:
I made a simple multi-object stl by exporting a platter. I can split it in the 2.4.58
can you give me a project where it fails to split?

jhlchu, 2022-04-16:
This is the one I used from the images above.
||Splits|
|-|-|
|SS 2.3.57|Yes|
|SS 2.4.58.12|No|
|SS 2.4 Nightly|Yes|
|PS 2.4.2|Yes|

[No split.zip](https://github.com/supermerill/SuperSlicer/files/8499139/No.split.zip)

supermerill, 2022-04-16:
if it's resolved in the nightly, great!

StephenRC, 2022-10-11:
Three objexts in one stl, made with openscad.