supermerill/SuperSlicer #2709: IDEX Question

inspectionsbybob, 2022-04-14:
### Version
2.3.57

### Operating system type + version
Win10

### Behavior
Questions about IDEX behavior in General...

I had forgotten how much of a Pain both IDEX and PVA are, and combine them and ouch...

Anyway...

Question on TOOL CHANGES...
If I allow the Idle print head to cool down to reduce oozing, is there a way to start the heating cycle BEFORE the Tool change?
I am looking at 5+ seconds of Warmup time per extruder per layer, and at 1,000 layers that hurts...

I saw something about starting the cool down on the wipe/retract, but was looking for the reverse....

Fighting some MARLIN IDEX issues as well.  It keeps pre-empting my custom Gcode, skipping parts and the like... also doing double retractions...

supermerill, 2022-04-14:
nop.

currently I have to create another gcode post-process for doing that.

As it's inefficient, difficult and dangerous, I prefer to wait for my eventual gcode writer rewrite that use symbolic object instead.

jeffs4fun, 2022-10-29:
i found a line in config adv.h
toolchange_no_return. 

this may help. 

it didnt pre heat for me but it did stay off to the side until it was heated.