supermerill/SuperSlicer #2714: Variable layer height doesn't work properly

Super-Matter, 2022-04-15:
### What happened?

The box shape is sliced correctly but my custom stl is not.

### Project file & How to reproduce

[image]
[image]
[image]

### Version

2.4.58.2

### Operating system

Win10

### Printer model

custom

Super-Matter, 2022-04-15:
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/8496902/Shape-Box.zip)
Here's my 3mf

Super-Matter, 2022-04-15:
[image]
When I scale it up it starts to work

Super-Matter, 2022-04-15:
[image]
I could trigger it working by adjusting the y scale factor only

Super-Matter, 2022-04-15:
2.3.57.12 seems to work. Ill stick to it for now. Can't compile a nightly build right now

supermerill, 2022-04-15:
you don't need to build, just use [download a good file](https://github.com/supermerill/SuperSlicer/suites/6065818425/artifacts/209601478)  from the action menu

Super-Matter, 2022-04-18:
Bugfix confirmed in 2.4.58.3 
Thanks

closed 2022-04-18 by Super-Matter