supermerill/SuperSlicer #2716: Autospeed defaults to 5mm/s when top or bottom layers are not 0

raarongray, 2022-04-15:
### Version
_Version of SuperSlicer used goes here (help->about)_

### Operating system type + version
Microsoft Windows 11 Home Version 10.0.22000 Build 22000_

### Behavior
* _Describe the problem_Using auto speed with 10mm^3/s flowrate causes drastically reduced speeds when there are top or bottom layers specified
* _Steps needed to reproduce the problem_Set Volumetric feedrate to 10mm^3/s and max speed to 80mm/s
* _Expected Results_ Speeds assigned to various features to maintain the fastest speeds without crossing the max volumetric feedrate
* _Actual Results_Speeds are barely above 5mm/s when any top or bottom layers are present
  * _Screenshots if useful_

_Is this a new feature request?_No
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here
[Autospeed.zip](https://github.com/supermerill/SuperSlicer/files/8498017/Autospeed.zip)
_

supermerill, 2022-04-15:
Look at gap fill speed tooltip.
note that their is now a pop-up warning in 2.4 just in case you didn't read it.

raarongray, 2022-04-16:
Thank you, I did not have gap fill enabled, so I kind of ignored it.