supermerill/SuperSlicer #2717: Compare presets => copy presets!

alexvonsiebenthal, 2022-04-17:
**Is your feature request related to a problem? Please describe.**
with new features being added (i.e. acceleration control), we need to update all profiles. It's a lot of work to do so manually.

**Describe the solution you'd like** / **Describe how it would work**
In the compare presets dialog, I'd like the ability to copy some presets from profile A to profile B.
[image]

**Describe alternatives you've considered**
Doing things manually

supermerill, 2022-04-17:
duplicate of #2524, i guess

alexvonsiebenthal, 2022-04-17:
> duplicate of #2524, i guess

Absolutely, sorry! I did not find it prior to creating this issue.

closed 2022-04-17 by alexvonsiebenthal