supermerill/SuperSlicer #2718: Missing speed setting for bridging perimeters

Gizzzle, 2022-04-17:
### What happened?

The current release of Superslicer is missing a setting to apply the bridging speed to the bridging perimeters, which leads to bad print quality in said areas. You can see the issue on the left airway (looking from behind) of the Voron Stealthburner: 

[image]

The current Prusaslicer has the option "Detect Bridging Perimeters" for exact this case.

### Project file & How to reproduce

[[a]_stealthburner_main_body_beta7.zip](https://github.com/supermerill/SuperSlicer/files/8501801/a._stealthburner_main_body_beta7.zip)

To see the said problem, go to layer 127.

### Version

2.4.58.2

### Operating system

Windows 10 64 Bit

### Printer model

Voron Trident 350

Dx-Dx-Dx, 2022-04-19:
[image]
This setting detects overhanging perimeters and applies bridge settings.
[image]

Gizzzle, 2022-04-19:
> [image] This setting detects overhanging perimeters and applies bridge settings. [image]

Oh, thanks for the hint! I really didn't associate this setting with my issue.

My problem seems to be solved now.