supermerill/SuperSlicer #2719: 2.4.58.2 crashes at startup on Mac Intel

LazaroFilm, 2022-04-17:
### What happened?

I click on the icon, then the app crashes. See crash report below

`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [84300]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.2_2022-03-31 (SuperSlicer_2.4.58.2_2022-03-31)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-04-17 10:51:03.6208 -0400
OS Version:            macOS 12.3.1 (21E258)
Report Version:        12
Bridge OS Version:     6.4 (19P4243)
Anonymous UUID:        A1CCECDA-CF24-F1F5-F8F9-9E059C24A8AA

Sleep/Wake UUID:       9A100FE5-8AA8-4801-BE69-7919D49ABDD8

Time Awake Since Boot: 93000 seconds
Time Since Wake:       6264 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file)
(terminated at launch; ignore backtrace)

Application Specific Information:
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file)

Thread 0 Crashed:
0   dyld                          	       0x10f02b0de __abort_with_payload + 10
1   dyld                          	       0x10f041212 abort_with_payload_wrapper_internal + 80
2   dyld                          	       0x10f041244 abort_with_payload + 9
3   dyld                          	       0x10efef76f dyld4::halt(char const*) + 375
4   dyld                          	       0x10efeb5a5 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3995
5   dyld                          	       0x10efea4d4 start + 388

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7bc088618  rdx: 0x00007ff7bc088a80
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7bc088660  rsp: 0x00007ff7bc088618
   r8: 0x00007ff7bc088680   r9: 0x0000000000000000  r10: 0x0000000000000075  r11: 0x0000000000000246
  r12: 0x0000000000000075  r13: 0x00007ff7bc088a80  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x000000010f02b0de  rfl: 0x0000000000000246  cr2: 0x000000010eff1e74
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x10efe5000 -        0x10f050fff dyld (*) <dd9e80de-fb3b-349b-96a4-46874ad34d11> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
[... 147 lines cut ...]
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Device: Thunderbolt Display, Apple Inc., 1, 26.2
`

### Project file & How to reproduce

Clicking on the app icon crashes every time

### Version

2.4.58.2

### Operating system

MacOs 12.3.1

### Printer model

HEVO

supermerill, 2022-04-22:
you need to install zstd for the current 2.4 build.
intall homebrew and then `brew install zstd`
see many 'macos'-tag issues on it here.

LazaroFilm, 2022-06-12:
Installing zstd solved it. Thanks!

closed 2022-06-12 by LazaroFilm